FreeCAD/FreeCAD #15870: coincident constraint not connecting line to a circle.

laytondavis, 2024-08-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Howto reproduce the bug:
Open Freecad, go into sketcher.
create a circle, then create a line with one end on the outside edge of the circle.
Freecad will automatically create a coincident constraint.
If you zoom in enough, you will see that the end of the line does not actually connect with the circle.

Example of where it works correctly:
Open Freecad, go into sketcher
create a rectangle, then create a line with one end on an edge of the rectangle
Freecad will automatically create a coincident constraint
you can zoom in as far as you are willing, and the point on the end of the line will always be on the edge of the rectangle.

When you export a DXF from a sketch containing the circle/line bug, and then import that DXF into other software, you always end up with broken geometry that will not work. For example, CAMBAM can not place milling operations using these geometries because they do not connect.

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-13:
Can you provide a demo file and maybe record a macro how you recreate this?

Roy-043, 2024-08-13:
The provided file is accurate. The endpoint of the line is 15.000000000000002 mm from the center of the circle. The radius of the circle is 15 mm. There is a very small rounding error, but such errors are normal when dealing with floating point numbers.

The 3D view is misleading as the circle is tessellated, it is represented as a polygon.

The problem must be how the DXF is created. There are many settings that can influence the result. My advice is to start a forum topic about that.

IMO this issue can be closed.

maxwxyz, 2024-08-13:
I think this is only a display issue with circular objects which are not rendered perfectly. The sketch is trimmable when adding a second edge and the profile is closed.
[image]

closed 2024-08-13 by maxwxyz