FreeCAD/FreeCAD #15871: Assembly: tangent joint is missing

FEA-eng, 2024-08-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Tangent joints are currently missing in Assembly and could be really useful. #15805 was requesting them but turned into a bug report related to Distance joint so I'm creating a new feature request.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-08-13:
Yes tangent is basically a distance of 0.
So I didn't want to add one tool to the toolbar. We could have one additional tool that would create a distance of 0.
I'd rather not add this but if people find this confusing... A topic for the DWG.

FEA-eng, 2024-08-13:
> Yes tangent is basically a distance of 0.

But can it work (keep tangency) e.g. in all the cases mentioned here: https://help.solidworks.com/2024/English/SolidWorks/sldworks/r_Tangent_Mate_SWassy.htm or is it somehow limited by the capabilities of the Distance joint?

maxwxyz, 2024-08-13:
I'm not sure. @PaddleStroke can distance 0 handle joints with references to lines / curves / circles and also arbitrary surfaces?

pierreporte, 2026-02-18:
I don’t know if this is the right issue, but I have tried the following things and it didn’t work. It’s quite blocking for what I do.

@PaddleStroke what do you thing of this?

------------

Here I am trying to make the end face of the cylinder tangent with the inner surface of the tube (or the outer surface) while having a joint so that the cylinder is coaxial with the tube.

[image]

--------------------------

Here it’s similar, I want to make the shoulder face tangent to the outer surface of the cylinder. there is a joint to make the part aligned with the hole.

[image]

----------------------------------

Here I’m trying to make the cylinder tangent to the side of the potato-shaped part. Basically it’s a cam roller, something very common in mechanism. It should also work with a ball.

[image]

[image]