FreeCAD/FreeCAD #15873: Tree view icon size preference does not work properly

Roy-043, 2024-08-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

You can change the Tree view icon size in the preferences. But only the icons of document nodes use that size. The other items use the default icon size, they only use the preference for their 'line height'.

https://wiki.freecad.org/Preferences_Editor#UI

[image]

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-13:
@kadet1090 FYI

kadet1090, 2024-08-13:
@Roy-043 do you think that simply removing that setting would be ok? Fixing it seems like waste of time.

Roy-043, 2024-08-16:
@MisterMakerNL Indeed it works if you enter a size that is smaller than the 'default' size. Entering a larger size does not work. In my case I want to increase the size from 16px to 32px (to create better screenshot for the Wiki).

It would be OK to removed this preference from the UI for now. But IMO this feature is important enough that it should be fixed. It is directly related to accessibility.

MisterMakerNL, 2024-08-16:
Bigger also works.
[image]

furgo16, 2024-10-10:
I can confirm. On Linux, neither increasing the size to 32 or 64 changes the icon sized of the elements in the tree, it only changes their line height. It does change the size of the top-level document icon in the tree, though.

> In my case I want to increase the size from 16px to 32px (to create better screenshot for the Wiki).

Funnily enough, I found myself in this same situation and I landed on this issue.

> It would be OK to removed this preference from the UI for now. But IMO this feature is important enough that it should be fixed. It is directly related to accessibility.

I'd agree on the second point. The accessibility factor should not be discounted.

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38926 (Git) Snap 1161
Build type: Release
Branch: main
Hash: a9bff78974f301100134a13c636e23aa91be5bcf
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

furgo16, 2024-10-10:
If it's not doable for 1.0, perhaps it can be added to a "[Known limitations](https://wiki.freecad.org/Release_notes_1.0#Known_Limitations)" section on the 1.0 release notes?

maxwxyz, 2026-11-07:
Part of this is fixed in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/3e27fbd30ee9fb3c1d466ba6e8d513edb912f520

Existing icons are refreshed when the size setting changes. Startup and default sizing aren't covered by this patch.