FreeCAD/FreeCAD #15875: Part Design: Pad and pocket functions are mixed up

Weeris102, 2024-08-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I want to pad a sketch which extends an existing pad. If I select the sketch and click on "Pad", it makes a pocket. If I do the opposite action, a pocket, it pads the sketch.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

PartDesign

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-08-13:
This usually means that the model is broken. Can you share the .FCStd file?

maxwxyz, 2024-08-13:
Please also record a macro when reproducing this. TIA.

BlenderManiac40, 2024-08-13:
I don't know, but on recent weekly build (38459) Pad and Pocket works as expected. Win 11 x64

Weeris102, 2024-08-14:
Thank for your replies. The problem was my additive and subtractive lofts.

closed 2024-08-14 by maxwxyz