FreeCAD/FreeCAD #15876: TechDraw: Exploded View from Assembly WB Problem

Pendelf, 2024-08-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When you add an Eploded View from Assembly WB to a sheet in TechDraw, it is not displayed correctly
[image]

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.24
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * OpenTheme 2024.7.3
  * PieMenu 1.7.6
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Assembly

### Anything else?

[https://forum.freecad.org/viewtopic.php?t=89411](url)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prokoudine, 2024-09-11:
The screenshot is not available. Could you please copy/paste as image (rather than link) it into the description?

Pendelf, 2024-09-11:
[image]

PaddleStroke, 2024-10-08:
I cannot reproduce this error.
Perhaps it's specific to your file. Can you please provide it?

luzpaz, 2024-10-14:
@PaddleStroke https://forum.freecad.org/download/file.php?id=271201

WandererFan, 2024-10-19:
If the new assembly WB was only implemented in the v0.22 devel cycle, how can this be a regression?

luzpaz, 2024-10-31:
@PaddleStroke JFYI :point_up_2:

PaddleStroke, 2024-11-04:
After some investigation, this bug happens only if you have both a view of the assembly and a view of the exploded-view.
If you have only a view of the explosion, then it works correctly.

PaddleStroke, 2024-11-07:
It appears that the problem is the way the case of exploded views are handled : it change the placements of the assembly component, extract the shape of the assembly, then reset the placements of the components.
Now in the case where there is also a simple view of the assembly I guess we are adding twice the assembly shape which kind of reset the exploded placements we have set during first extraction. Or something like this.

Fortunately I found an easy fix

Pendelf, 2024-11-19:
Still not working properly

https://github.com/user-attachments/assets/925dc29d-9375-4420-b7bb-2a64cf4da33e

[image]

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.30
  * PieMenu 1.9.5
  * sheetmetal 0.5.5

PaddleStroke, 2024-11-19:
Ah damn I can confirm that this is not fixed. I don't know why it somehow worked when I was fixing this but clearly it is not working.

luzpaz, 2024-11-20:
Re-opened ticket

WandererFan, 2024-12-14:
Another forum post: https://forum.freecad.org/viewtopic.php?t=92965

lavalade, 2024-12-14:
I have a similar problem: the exploded view ok, but messed up in techdraw, see the two screenshots.
See my [original post on forum.freecad.org](https://forum.freecad.org/viewtopic.php?t=92965) for full details and the `*.FCStd` file.

[image]
[image]

SirMew, 2024-12-14:
I'm having a similar problem. I have two techdraw pages, one for the regular assembly and another for the exploded view. 

https://github.com/user-attachments/assets/27791270-00c8-4636-885e-88ecda24e36f

SE-Hans, 2024-12-15:
Hello I have the same problem.
After a few attempts I found out that not the "Exploded_View" is used, but the originals **with the coordinates as they were created** in PartDesign.
If you move the originals, they will be displayed in their new position with the Exploded_Views rays, if you paste Exploded_Views -> TechDraw repeatedly.
I have done experiments with: "All parts in one file", "Parts and assembly in two separate files" and "different parts in separate files and assembly in their own file". 
Every attempt can be verified. 
In the beginning, I followed the procedure of (https://www.youtube.com/watch?v=k51YFdXIIek) exactly and then tried various others.
I'll spare you sample pictures, because everything has already been shown above and it would be a lot. 
But the result is always the same :-(

I tested on Ubuntu 24.04.1 and Windows 10 on laptop, barebone and desktop PC with the latest Freecad 1.0 version.

Maybe my knowledge will help to fix it. 

Best regards
Hans

p.s.: The text is machine translated, sorry ;-)

PaddleStroke, 2024-12-15:
I need to review this. My previous PR made things worse actually. At least before if you had only the exploded view it was working.

SE-Hans, 2024-12-16:
I have always used only the exploded, never a second/other.

SirMew, 2024-12-16:
I tried deleting the other page and doing just an exploded view - same issue :/

Pendelf, 2024-12-18:
I came across this, when activating Exploded View, it fails to deactivate the assembly, it helps to close the file and reopen it. This may be related to the current problem

https://github.com/user-attachments/assets/c2e93531-6841-43a6-8d85-fc4f04816316

PaddleStroke, 2024-12-19:
@Pendelf this is a different issue. I confirm that I can reproduce it. Can you open an issue without your video please?

Pendelf, 2024-12-19:
@PaddleStroke I don't quite get it. Do I need to open an issue on this topic or is there something wrong with the video?

Sairus1, 2025-02-22:
Hello

I was following a tutorial, and could achieve the explode parts, but when I want to insert in a Tech Draw, the parts appears all in the origin, in the same position that it was sketched and paded.
Thank for your patient, I'm learning and noob.

toenails6, 2025-06-10:
Had this issue, then when I edit the exploded view in the assembly again, and click on okay instead of cancel, then recompute techdraw view object, the exploded view updated into the drawing. However, I have multiple exploded view, and somehow other views all show the same exploded view despite having different exploded views as the source.

luzpaz, 2025-07-09:
bumping for visibility

PaddleStroke, 2025-10-17:
unfortunately not yet. I can check it out again but the problem was quite complex