FreeCAD/FreeCAD #15879: TNP when deleting a feature

maxwxyz, 2024-08-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Not sure if it is the same as: https://github.com/FreeCAD/FreeCAD/issues/14643
When recreating a different issue I've noticed a TNP bug which throws an error when deleting a prior feature of a fillet but the fillet has the correct edge in it when opening. When opening the fillet dialog and closing it again with OK the feature is hidden too.
So the correct edge is actually in the feature but I need to open and close the feature again that it works.

This is the error:
[image]

Demo:

https://github.com/user-attachments/assets/45bfdb35-08be-4378-b75f-65ed0fa380ca

The error also occurs when I use another fillet instead of a pad and delete it:

https://github.com/user-attachments/assets/6c87573d-e82a-463d-add1-f7453c39eb54

Here is the example file:
[Loft-v22.zip](https://github.com/user-attachments/files/16609516/Loft-v22.zip)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

PartDesign

### Anything else?

Macro, starting with the open example file:

```
# -*- coding: utf-8 -*-

# Macro Begin: C:\Users\max\AppData\Roaming\FreeCAD\Macro\TNP-issue.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++
import FreeCAD
import PartDesign

# Gui.runCommand('Std_DlgMacroRecord',0)
# Gui.Selection.addSelection('Loft_v22','Body')
# Gui.ActiveDocument.ActiveView.setActiveObject('pdbody',App.getDocument('Loft_v22').getObject('Body'))
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Loft_v22','Body','AdditiveLoft.Edge2',-8.90375,4.55229,15)
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Loft_v22','Body','AdditiveLoft.Face1',-9.88142,5.33777,15)
### Begin command PartDesign_Pad
App.getDocument('Loft_v22').getObject('Body').newObject('PartDesign::Pad','Pad')
App.getDocument('Loft_v22').getObject('Pad').Profile = (App.getDocument('Loft_v22').getObject('AdditiveLoft'), ['Face1',])
App.getDocument('Loft_v22').getObject('Pad').Length = 10
App.ActiveDocument.recompute()
App.ActiveDocument.recompute()
# App.getDocument('Loft_v22').getObject('Pad').ViewObject.ShapeAppearance=getattr(App.getDocument('Loft_v22').getObject('AdditiveLoft').getLinkedObject(True).ViewObject,'ShapeAppearance',App.getDocument('Loft_v22').getObject('Pad').ViewObject.ShapeAppearance)
# App.getDocument('Loft_v22').getObject('Pad').ViewObject.LineColor=getattr(App.getDocument('Loft_v22').getObject('AdditiveLoft').getLinkedObject(True).ViewObject,'LineColor',App.getDocument('Loft_v22').getObject('Pad').ViewObject.LineColor)
# App.getDocument('Loft_v22').getObject('Pad').ViewObject.PointColor=getattr(App.getDocument('Loft_v22').getObject('AdditiveLoft').getLinkedObject(True).ViewObject,'PointColor',App.getDocument('Loft_v22').getObject('Pad').ViewObject.PointColor)
# App.getDocument('Loft_v22').getObject('Pad').ViewObject.Transparency=getattr(App.getDocument('Loft_v22').getObject('AdditiveLoft').getLinkedObject(True).ViewObject,'Transparency',App.getDocument('Loft_v22').getObject('Pad').ViewObject.Transparency)
# App.getDocument('Loft_v22').getObject('Pad').ViewObject.DisplayMode=getattr(App.getDocument('Loft_v22').getObject('AdditiveLoft').getLinkedObject(True).ViewObject,'DisplayMode',App.getDocument('Loft_v22').getObject('Pad').ViewObject.DisplayMode)
# Gui.getDocument('Loft_v22').setEdit(App.getDocument('Loft_v22').getObject('Body'), 0, 'Pad.')
# Gui.Selection.clearSelection()
### End command PartDesign_Pad
# Gui.Selection.clearSelection()
App.getDocument('Loft_v22').getObject('Pad').Length = 13.000000
App.getDocument('Loft_v22').getObject('Pad').TaperAngle = 0.000000
App.getDocument('Loft_v22').getObject('Pad').UseCustomVector = 0
App.getDocument('Loft_v22').getObject('Pad').Direction = (0, 0, 1)
App.getDocument('Loft_v22').getObject('Pad').ReferenceAxis = None
App.getDocument('Loft_v22').getObject('Pad').AlongSketchNormal = 1
App.getDocument('Loft_v22').getObject('Pad').Type = 0
App.getDocument('Loft_v22').getObject('Pad').UpToFace = None
App.getDocument('Loft_v22').getObject('Pad').Reversed = 1
App.getDocument('Loft_v22').getObject('Pad').Midplane = 0
App.getDocument('Loft_v22').getObject('Pad').Offset = 0
App.getDocument('Loft_v22').recompute()
App.getDocument('Loft_v22').getObject('AdditiveLoft').Visibility = False
# Gui.getDocument('Loft_v22').resetEdit()
App.getDocument('Loft_v22').getObject('AdditiveLoft').Visibility = False
# Gui.Selection.addSelection('Loft_v22','Body','Pad.Edge1',11.565,4.74338,15)
### Begin command PartDesign_Fillet
App.getDocument('Loft_v22').getObject('Body').newObject('PartDesign::Fillet','Fillet')
App.getDocument('Loft_v22').getObject('Fillet').Base = (App.getDocument('Loft_v22').getObject('Pad'),['Edge1',])
# Gui.Selection.clearSelection()
App.getDocument('Loft_v22').getObject('Pad').Visibility = False
App.ActiveDocument.recompute()
# App.getDocument('Loft_v22').getObject('Fillet').ViewObject.ShapeAppearance=getattr(App.getDocument('Loft_v22').getObject('Pad').getLinkedObject(True).ViewObject,'ShapeAppearance',App.getDocument('Loft_v22').getObject('Fillet').ViewObject.ShapeAppearance)
# App.getDocument('Loft_v22').getObject('Fillet').ViewObject.LineColor=getattr(App.getDocument('Loft_v22').getObject('Pad').getLinkedObject(True).ViewObject,'LineColor',App.getDocument('Loft_v22').getObject('Fillet').ViewObject.LineColor)
# App.getDocument('Loft_v22').getObject('Fillet').ViewObject.PointColor=getattr(App.getDocument('Loft_v22').getObject('Pad').getLinkedObject(True).ViewObject,'PointColor',App.getDocument('Loft_v22').getObject('Fillet').ViewObject.PointColor)
# App.getDocument('Loft_v22').getObject('Fillet').ViewObject.Transparency=getattr(App.getDocument('Loft_v22').getObject('Pad').getLinkedObject(True).ViewObject,'Transparency',App.getDocument('Loft_v22').getObject('Fillet').ViewObject.Transparency)
# App.getDocument('Loft_v22').getObject('Fillet').ViewObject.DisplayMode=getattr(App.getDocument('Loft_v22').getObject('Pad').getLinkedObject(True).ViewObject,'DisplayMode',App.getDocument('Loft_v22').getObject('Fillet').ViewObject.DisplayMode)
# Gui.getDocument('Loft_v22').setEdit(App.getDocument('Loft_v22').getObject('Body'), 0, 'Fillet.')
# Gui.Selection.clearSelection()
### End command PartDesign_Fillet
# Gui.Selection.clearSelection()
App.getDocument('Loft_v22').getObject('Fillet').Radius = 1.000000
App.getDocument('Loft_v22').getObject('Fillet').Base = (App.getDocument('Loft_v22').getObject('Pad'),["Edge1",])
App.getDocument('Loft_v22').recompute()
App.getDocument('Loft_v22').getObject('Pad').Visibility = False
# Gui.getDocument('Loft_v22').resetEdit()
# Gui.Selection.addSelection('Loft_v22','Body','Pad.')
### Begin command Std_Delete
App.getDocument('Loft_v22').getObject('Body').removeObject(App.getDocument('Loft_v22').getObject('Pad'))
App.getDocument('Loft_v22').removeObject('Pad')
App.getDocument('Loft_v22').recompute()
### End command Std_Delete
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Loft_v22','Body','Fillet.')
# Gui.ActiveDocument.setEdit(App.getDocument('Loft_v22').getObject('Fillet'), 0)
# Gui.Selection.clearSelection()
App.getDocument('Loft_v22').getObject('Fillet').Radius = 1.000000
App.getDocument('Loft_v22').getObject('Fillet').Base = (App.getDocument('Loft_v22').getObject('AdditiveLoft'),["Edge1",])
App.getDocument('Loft_v22').recompute()
App.getDocument('Loft_v22').getObject('AdditiveLoft').Visibility = False
# Gui.getDocument('Loft_v22').resetEdit()
# Gui.Selection.addSelection('Loft_v22','Body','Fillet.')
# Gui.Selection.clearSelection()
# Macro End: C:\Users\max\AppData\Roaming\FreeCAD\Macro\TNP-issue.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++

```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-14:
@bgbsww is this actually the same as https://github.com/FreeCAD/FreeCAD/issues/14643 because it throws the error when deleting a feature but when opening the feature in the task dialog, the correct edges are displayed and just clicking OK solves the issue.

bgbsww, 2024-08-15:
Okay, I'm super confused here:  It looks to me like you are making a pad on top of the loft, and then making a fillet on an edge of that pad:
`App.getDocument('Loft_v22').getObject('Fillet').Base = (App.getDocument('Loft_v22').getObject('Pad'),['Edge1',])`
You then delete that pad, leaving the fillet with nothing to apply to - which it warns about.
That all seems reasonable, I guess I'm missing something.
Then, you open the fillet, and it takes a stab and connects it to a remaining edge of the additive loft:
`App.getDocument('Loft_v22').getObject('Fillet').Base = (App.getDocument('Loft_v22').getObject('AdditiveLoft'),["Edge1",])
` - a totally different object

The two different edges - the one in Pad that got deleted, and the one in the additive loft coincide in space and in history of how they came to be, so the code is guessing a substitution here that you are approving in that fillet dialog,

Which part of this are we saying should be happening, or should not be happening?

maxwxyz, 2024-08-15:
@bgbsww It chooses the correct edge after the pad is deleted (actually the pad does not create new geometry, as auto refine is on and I padded it inside the existing body). It was Edge1 when the pad was there, after deleting the pad I get warnings and the fillet is gone, but when opening the fillet dialog and just close it again it works again (as it still has Edge1 in it). So why did it break if it has the right info stored?

Here is a simpler example with two fillets and no pad but the fillet does not find the correct edge but just opening and closing the fillet dialog repairs it but then hides the feature which need to be activated (it somehow messes with visibility too):

Start file (hint: it works only with this lofted part, a padded sketch which produces the same geometry will destroy the fillet completely):
[fillet-start.zip](https://github.com/user-attachments/files/16622323/fillet-start.zip)

Demo video:

https://github.com/user-attachments/assets/fe34a059-462a-4e67-81aa-a55069c492d3

And the recorded macro:
```
# -*- coding: utf-8 -*-

# Macro Begin: C:\Users\max\AppData\Roaming\FreeCAD\Macro\tnp-additive-loft.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++
import FreeCAD
import PartDesign

# Gui.runCommand('Std_DlgMacroRecord',0)
# Gui.Selection.addSelection('Unnamed1','Body','AdditiveLoft.Edge1',14.2401,14.0435,10)
### Begin command PartDesign_Fillet
App.getDocument('Unnamed1').getObject('Body').newObject('PartDesign::Fillet','Fillet')
App.getDocument('Unnamed1').getObject('Fillet').Base = (App.getDocument('Unnamed1').getObject('AdditiveLoft'),['Edge1',])
# Gui.Selection.clearSelection()
App.getDocument('Unnamed1').getObject('AdditiveLoft').Visibility = False
App.ActiveDocument.recompute()
# App.getDocument('Unnamed1').getObject('Fillet').ViewObject.ShapeAppearance=getattr(App.getDocument('Unnamed1').getObject('AdditiveLoft').getLinkedObject(True).ViewObject,'ShapeAppearance',App.getDocument('Unnamed1').getObject('Fillet').ViewObject.ShapeAppearance)
# App.getDocument('Unnamed1').getObject('Fillet').ViewObject.LineColor=getattr(App.getDocument('Unnamed1').getObject('AdditiveLoft').getLinkedObject(True).ViewObject,'LineColor',App.getDocument('Unnamed1').getObject('Fillet').ViewObject.LineColor)
# App.getDocument('Unnamed1').getObject('Fillet').ViewObject.PointColor=getattr(App.getDocument('Unnamed1').getObject('AdditiveLoft').getLinkedObject(True).ViewObject,'PointColor',App.getDocument('Unnamed1').getObject('Fillet').ViewObject.PointColor)
# App.getDocument('Unnamed1').getObject('Fillet').ViewObject.Transparency=getattr(App.getDocument('Unnamed1').getObject('AdditiveLoft').getLinkedObject(True).ViewObject,'Transparency',App.getDocument('Unnamed1').getObject('Fillet').ViewObject.Transparency)
# App.getDocument('Unnamed1').getObject('Fillet').ViewObject.DisplayMode=getattr(App.getDocument('Unnamed1').getObject('AdditiveLoft').getLinkedObject(True).ViewObject,'DisplayMode',App.getDocument('Unnamed1').getObject('Fillet').ViewObject.DisplayMode)
# Gui.getDocument('Unnamed1').setEdit(App.getDocument('Unnamed1').getObject('Body'), 0, 'Fillet.')
# Gui.Selection.clearSelection()
### End command PartDesign_Fillet
# Gui.Selection.clearSelection()
App.getDocument('Unnamed1').getObject('Fillet').Radius = 1.000000
App.getDocument('Unnamed1').getObject('Fillet').Base = (App.getDocument('Unnamed1').getObject('AdditiveLoft'),["Edge1",])
App.getDocument('Unnamed1').recompute()
App.getDocument('Unnamed1').getObject('AdditiveLoft').Visibility = False
# Gui.getDocument('Unnamed1').resetEdit()
# Gui.Selection.addSelection('Unnamed1','Body','Fillet.Edge2',4.42808,8.96616,10)
### Begin command PartDesign_Fillet
App.getDocument('Unnamed1').getObject('Body').newObject('PartDesign::Fillet','Fillet001')
App.getDocument('Unnamed1').getObject('Fillet001').Base = (App.getDocument('Unnamed1').getObject('Fillet'),['Edge2',])
# Gui.Selection.clearSelection()
App.getDocument('Unnamed1').getObject('Fillet').Visibility = False
App.ActiveDocument.recompute()
# App.getDocument('Unnamed1').getObject('Fillet001').ViewObject.ShapeAppearance=getattr(App.getDocument('Unnamed1').getObject('Fillet').getLinkedObject(True).ViewObject,'ShapeAppearance',App.getDocument('Unnamed1').getObject('Fillet001').ViewObject.ShapeAppearance)
# App.getDocument('Unnamed1').getObject('Fillet001').ViewObject.LineColor=getattr(App.getDocument('Unnamed1').getObject('Fillet').getLinkedObject(True).ViewObject,'LineColor',App.getDocument('Unnamed1').getObject('Fillet001').ViewObject.LineColor)
# App.getDocument('Unnamed1').getObject('Fillet001').ViewObject.PointColor=getattr(App.getDocument('Unnamed1').getObject('Fillet').getLinkedObject(True).ViewObject,'PointColor',App.getDocument('Unnamed1').getObject('Fillet001').ViewObject.PointColor)
# App.getDocument('Unnamed1').getObject('Fillet001').ViewObject.Transparency=getattr(App.getDocument('Unnamed1').getObject('Fillet').getLinkedObject(True).ViewObject,'Transparency',App.getDocument('Unnamed1').getObject('Fillet001').ViewObject.Transparency)
# App.getDocument('Unnamed1').getObject('Fillet001').ViewObject.DisplayMode=getattr(App.getDocument('Unnamed1').getObject('Fillet').getLinkedObject(True).ViewObject,'DisplayMode',App.getDocument('Unnamed1').getObject('Fillet001').ViewObject.DisplayMode)
# Gui.getDocument('Unnamed1').setEdit(App.getDocument('Unnamed1').getObject('Body'), 0, 'Fillet001.')
# Gui.Selection.clearSelection()
### End command PartDesign_Fillet
# Gui.Selection.clearSelection()
[... cut ...]
App.getDocument('Unnamed1').getObject('Fillet001').Radius = 1.000000
App.getDocument('Unnamed1').getObject('Fillet001').Base = (App.getDocument('Unnamed1').getObject('AdditiveLoft'),["Edge2",])
App.getDocument('Unnamed1').recompute()
App.getDocument('Unnamed1').getObject('AdditiveLoft').Visibility = False
# Gui.getDocument('Unnamed1').resetEdit()
# Gui.Selection.addSelection('Unnamed1','Body','Fillet001.')
# Macro End: C:\Users\max\AppData\Roaming\FreeCAD\Macro\tnp-additive-loft.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++
```

bgbsww, 2024-08-15:
Fillet001 is referring to Edge1 of Fillet -  Fillet.Edge1.  You then delete fillet, after being warned that doing this is going to break a reference.  That reference then breaks.  You then open Fillet001, and _at dialog open it automagically guesses that the completely different AdditiveLoft.Edge1 is a close match to what you intended_ and **suggests** it by putting it in the dialog and display.  You approve this change, and life goes on.  But  Fillet.Edge1 and AdditiveLoft.Edge1 are two different things that happen share common history and geometry, thus the TNP history match does the right thing.

I am still confused as to what the problem is here - what is happening that you do not expect?
 
The fillet is breaking because it is on an object that you are deleting.

Are you looking for the guessing to just act without checking in with the user?  I do think this is a reasonable future capability to look into in some cases, but not what we're doing right now.  The heuristic would likely need to make sure this happens when there is only one possibility to choose, and might want to happen before the initial dialog warning that you're about to delete a reference - "Warning, deleting this object may cause some references to break, but I know what those references are and have successfully guessed how to replace them by referring to identical geometry on another object - okay to proceed?"

maxwxyz, 2024-08-16:
Ok then I misunderstood what TNP fixes should solve. I thought as the same edge is present from the base feature, any feature added later which does not touch/modify this edge would keep pointing to the edge so it would not break even when adjusting / deleting features in between.
As I understand it, this would also mean, when adjusting the fillet feature in between to increase it's radius that some new edges are generated or even add/remove new edges the model after this feature will break. If not I'd expect the same as if the fillet is deleted.
But the opposite is the case. I can easily add/remove edges from the second filet feature without breaking it, so why will deleting this feature destroy it (it's the same as if the edge was removed from it).

bgbsww, 2024-08-16:
Yeah, we are disagreeing on what "the same edge" means.  Visually and geometrically, it is clearly the same edge, and thus the user expectation ( what you are noticing ).  But from a document history perspective, it's a totally different edge because it's on a different object.

Changing the radius of a fillet would doesn't necessarily change the number of edges and wouldn't have an effect.

Also, if the object still exists, then it becomes possible to find and correct the references ( mitigate the TNP ) for various edge changes.  It is the delete operation, where there is nothing left to search in that causes the issue.

By analogy:  Assume there's a building across the street with one circular window.  You talk to people about the circular window and they understand you.  If you add or subtract more rectangular windows from the building, everyone still understands what you're talking about regardless of whether the circular window is "the first window" or "the third window" etc - the reference is solid.

Now destroy the building.  You can talk all you want about the circular window, and people will think you are crazy because there is no building there at all to reference.

maxwxyz, 2024-08-16:
Would it make sense to label this as feature request then or is this not solvable at all?
I thought maybe look at the edge, if the current feature did not modify/create it, keep the reference to the prior feature, all the way back to what created this edge. Then it would not get lost on deletion.

bgbsww, 2024-08-16:
Feature request makes sense to me.  A plane / shape geometry matching
heuristic would work here.  I think there's a UI question here - should we
be automatically moving to an edge on a different object, or inform the
user, or ask the user if we should do it, etc.

maxwxyz, 2024-08-16:
@FreeCAD/design-working-group IMO we should try to automatically match the geometry and if this is not 100% possible, ask the user to pick the geometry instead of breaking all features and manually opening them all one by one, deleting the wrong geometry and adding the correct one.

pierreporte, 2024-08-16:
@maxwxyz I agree with you. If FreeCAD can determine with absolute certainty the right geometry to apply features on, then the change should be automatic. Otherwise, the user should be proposed the best guess, to be accepted, modified, or discarded entirely.

bgbsww, 2024-08-17:
I also agree that a dialog proposing changes makes sense.  That's a new feature, and so post-1.0, so I'm deassigning for now.

maxwxyz, 2025-02-04:
Could be the same as #14643

drwho495, 2025-03-20:
having this issue aswell