FreeCAD/FreeCAD #15881: project created with v0.21 can't be edited with v0.22

Panda631, 2024-08-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Find attached a file created with v0.21. Everything is OK.

If you open it in v0.22, the body is visible, but a number of error messages are displayed.

[image]

The check geometry result doesn't show any problem.

[image]

After recompute, the body is destroyed.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * EM 2.1.1
  * fasteners 0.5.21
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.1.23
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Anything else?

[StativKopf_2.zip](https://github.com/user-attachments/files/16613049/StativKopf_2.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-14:
I can confirm.
@bgbsww could this be related to TNP changes for the mirror command?
@NomAnor could this be related to your changes?

MarcusMWilliams, 2024-08-14:
> @maxwxyz <https://github.com/maxwxyz> looks like mirror and other
> transforms had been broken in 2b97b20
> and fixed in a1ce983

Yes mirror was fixed

bgbsww, 2024-08-14:
Can someone duplicate the "body is destroyed on recompute"?  When I try it, it just repairs the file and operates normally.

amon-sha, 2024-08-14:
@bgbsww see https://github.com/FreeCAD/FreeCAD/issues/15881#issuecomment-2288905617

maxwxyz, 2024-08-14:
@Panda631 So test with the next weekly on Friday please and if the issue is still present report back and we reopen this issue.

closed 2024-08-14 by maxwxyz

Panda631, 2024-08-17:
> @Panda631 So test with the next weekly on Friday please and if the issue is still present report back and we reopen this issue.

After loading the model, the error messages are still present. The geometry looks OK, also after recomputing.

When I save the file and open it again, the errors have disappeared. I would say problem solved.