FreeCAD/FreeCAD #15882: Selecting items in sketch does not result in visual indication

LHC-Canada, 2024-08-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Bear with me – this is the first time I have reported anything on git !

I have a sketch that was created from a step file using draft 2dshape and then draft convert to sketch.

I deleted a bunch of geometry and did nothing else to the sketch.  It’s unattached to anything and I have not added any constraints.

The sketch consists of nothing but circles and when I open it, I can select circles but only one of the circles (the center top one) will show it has been selected by turning green.  

The other circles are selectable, but they do not change color to indicate this.  I can for example, select two circles via their center reference points and then add a constraint such as vertical, horizontal, and the constraint is applied.

I have tried opening this sketch with dev versions 38419, 38429 and 38459

There is an error that appears when the file is opened that reads -
Reading Failed From Embedded File: StringHasher.Table.txt

File Attached

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38429 (Git)
Build type: Release
Branch: main
Hash: 362cf1c80edb89957d81c0dbe6e50ce35daaa6db
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.56a
  * Beltrami 1.0.8-alpha
  * Curves 0.4.4
  * Defeaturing
  * ExplodedAssembly
  * fasteners 0.3.44
  * fcgear 1.0.0
  * IconThemes
  * kicadStepUpMod 10.13.0
  * Manipulator 1.4.5
  * QuickMeasure 2022.10.28
  * scripts
  * sheetmetal 0.4.11
  * VendorParts
```

### Subproject(s) affected?

Sketcher

### Anything else?

[SketcherSelectionIssue.zip](https://github.com/user-attachments/files/16613607/SketcherSelectionIssue.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-14:
I cannot reproduce:
[image]

Please try the latest dev version from this week and recompute the object.
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

LHC-Canada, 2024-08-14:
Select the center points not the circle and see if you can reproduce then.  I am using the latest weekly dev version now (38450) and it still acts as described.  Thanks.

maxwxyz, 2024-08-14:
Yes I can reproduce but when drawing new circle it works, so could be a corrupted file?

LHC-Canada, 2024-08-14:
Quite possibly - but the corruption might still be caused by FreeCad as the original file that was imported was a step file.  I am going to go back and see if I can retrace my steps and see if I get anything different.  Might be a day or so.   I can rip the original step import back to a single object I believe.  Then I will do the draft 2dshape followed by the draft convert to sketch.  That's the sequence that I followed before.

LHC-Canada, 2024-08-14:
ok - more investigations done.  I brought in the original step file and reduced it to one object.  I then created a draft 2dshape and from that a draft convert to sketch.  I did no editing to the sketch.  Here are some pics of the weirdness.  Look at the center hole patterns - there is a stepper motor hole pattern and then the original hole pattern I filed the issue with.  If I edit the sketch and then window on the pattern on the right of the stepper motor pattern, only a couple of the center reference points show up but ALL of the circles are highlighted as being selected.

If I then just window on the stepper motor pattern ALL of the center reference points are highlighted as well as the circles.

If I then highlight the stepper motor pattern and delete it, then go and highlight the other pattern to the right, a lot more center reference points are highlighted as selected but not all of them.

Certainly looks like something is being corrupted, and freecad is altering what it highlights depending on what other geometry has been removed prior.....
[image]
[image]
[image]
[image]
[TestSketchSelection.zip](https://github.com/user-attachments/files/16619273/TestSketchSelection.zip)

LHC-Canada, 2024-08-14:
Perhaps there is something wrong in the single object that I am starting with ?  I have no clue what it could be or even how to check for it though.  I'm just a "regular enthusiastic FreeCad user" and not a developer or code guy.

maxwxyz, 2024-08-15:
@PaddleStroke @FlachyJoe any idea?

maxwxyz, 2024-09-01:
@bgbsww could this be related to https://github.com/FreeCAD/FreeCAD/issues/16135 mechanics?

bgbsww, 2024-09-01:
Congrats on your first bug submission.  Keep it up if you can find any more [joke].

Yes, this is absolutely the same issue as #16135 and a fix is about to be placed in a PR.

LHC-Canada, 2024-09-01:
Awesome !  Will keep an eye out and check on the next weekly dev version.  Happy I could help FreeCad in some small way.

PR #16203 (2024-09-02): Toponaming: Fix ctrl selection issues by looking up element names and…
… setting up detail path.  Fix #16135.  Fix #15882.