FreeCAD/FreeCAD #15883: Multitransform is broken.

leoheck, 2024-08-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

- Features disappear from the list.
- I can´t add them anymore. Buttons are disabled.
- What is the difference between "Transform body" and "Transform tools shapes"? Both are doing the same. 
[image]

Also, the second option is selected by default. This is crazy. Please, select the first one as the default, instead. 
If the "Transform tools shapes" should be the default, move it up to the list, please.
[image]

Also, the operation is broken.
See, the red should be possible but you cannot see here, since Adding the Feature ***STILL HIDES*** the whole BODY!! 
[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git) AppImage
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.24
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.8.10
  * fasteners 0.5.26
  * asm4_disable_version_check_popup
  * OpticsWorkbench 1.0.20
  * lattice2 1.0.0
  * kicadStepUpMod
  * OpticsWorkbench.backup1722883583.3368473 1.0.18 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leoheck, 2024-08-14:
If I "transform the BODY" which is not the "Default" operation, it seems to work. 
[image]

So what does it mean the options? Any tooltip for the end user?
Why the second one is selected by default?
If the second should be the default, please, move it to the top of the list.

leoheck, 2024-08-14:
When I edit the Multitransofrm Feature, the base feature disapears.

maxwxyz, 2024-08-14:
@amon-sha @bgbsww is this also fixed by https://github.com/FreeCAD/FreeCAD/commit/a1ce9830358129cda4d3d2cf66a803aa22e13429 if so, I'll close this.

bgbsww, 2024-08-14:
I think so; the version should be .38461 or later to include the fix, so it would be useful to confirm we don't see the issue on those versions.  The bug description is certainly consistent with what i think I fixed.

closed 2024-08-15 by maxwxyz