FreeCAD/FreeCAD #15888: ViewProviderPythonFeature should be renamed ViewProviderFeaturePython

aiksiongkoh, 2024-08-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

ViewProviderPythonFeature should be renamed ViewProviderFeaturePython to be consistent with FeaturePython,
PythonFeature does not exist.

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.65
  * fasteners 0.5.24
  * scripts
  * TemplatePyMod
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-15:
Please make sure that you're checking the development build instead of stable.  
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

aiksiongkoh, 2024-08-15:
https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/freecad_source.tar.gz
still has the error as of today.

luzpaz, 2024-08-17:
@xtemp09 Do you mind weighing-in ?

xtemp09, 2024-08-17:
@luzpaz, what do you mean by "weigh-in?" Here is a line from the file mentioned:

https://github.com/FreeCAD/FreeCAD/blob/85cbd31389dcceff797f4a8c0a8a50ee8622453b/src/Gui/ViewProviderPythonFeature.cpp#L2

Or other lines from other files:

https://github.com/FreeCAD/FreeCAD/blob/85cbd31389dcceff797f4a8c0a8a50ee8622453b/src/App/FeaturePython.h#L162-L166

or here:

https://github.com/FreeCAD/FreeCAD/blob/85cbd31389dcceff797f4a8c0a8a50ee8622453b/src/App/FeaturePython.cpp#L71-L74

luzpaz, 2024-08-17:
@xtemp09 thanks, you did exactly the thing I was hoping. :pray: 

CC @wwmayer :point_up_2: what say you ?

wwmayer, 2024-08-18:
OK. See https://github.com/FreeCAD/FreeCAD/pull/15940

PR #15940 (2024-08-26): Core: Rename ViewProviderPythonFeature to ViewProviderFeaturePython
Fixes #15888
commit 3e33f184 (2024-08-26): Core: Rename ViewProviderPythonFeature to ViewProviderFeaturePython

Fixes #15888
closed 2024-08-26 by yorikvanhavre