FreeCAD/FreeCAD #15889: Unable to select edge when second time to chamfer or fillet the model

LEWeihuaV, 2024-08-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

# Describe
Unable to select edge when second time to chamfer or fillet the model.
# Picture
## chamfer

[image]
## fillet
[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Chinese/China (zh_CN)
[/code]
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-15:
Could you please try this in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds and please add steps how to reproduce and a demo file.

LEWeihuaV, 2024-08-16:
of course，the problem of this report is appearing when i want to fillet or charfer  the model in second time and i have recorded a video if you want to check it i could send it by email.On the other hand,it have a normal performance when i operation(Edit) repeatedly in first function icon of fillet or charfer,instead of create a new function icon with fillet.I don't konw the performance like that is okay!

maxwxyz, 2024-08-16:
Please test on the dev version. You can record a macro and upload it here and you can also upload a video directly to GitHub and also a zipped FCStd file.

closed 2024-08-19 by LEWeihuaV