FreeCAD/FreeCAD #15890: Cannot switch to Classic theme

maxwxyz, 2024-08-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The latest weekly broke the ability to switch to the Classic theme at all. Switching between Light and Dark works normally.

Demo:

https://github.com/user-attachments/assets/d97858e2-ac8c-4e0b-8584-69b71dd15870

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-15:
@Syres916 could this be a regression of your https://github.com/FreeCAD/FreeCAD/pull/15687 PR

Syres916, 2024-08-15:
Most probably, it works fine on Linux but please ask one of the maintainers to revert https://github.com/FreeCAD/FreeCAD/commit/2995e6aeb762991fbb04647b010f0a897e51daae unfortunately I've not got the required time to investigate for at least a week.

bdieterm, 2024-08-15:
I was able to reproduce the problem on Linux:
Switching to the dark theme, restarting FreeCAD and then switching back to the classic theme still kept the dark style.
In user.cfg file I have seen that entry "StyleSheet" was not reset to the value for the classic theme.

dark theme:
`<FCText Name="StyleSheet">FreeCAD Dark.qss</FCText>`

classic theme:
`<FCText Name="StyleSheet"></FCText>`

After applying the changes from #15894, switching back to classic style works again (not dark anymore).

chennes, 2024-08-18:
@obelisk79 is going to put together a more complete Preference Pack config file here, so that we explicitly set these things to their defaults. Future work (post-1.0) will include loading that pack on first start, so that it is the single authoritative source for all style-related defaults (with the defaults provided in code during parameter access purely vestigial).

closed 2024-08-31 by chennes