FreeCAD/FreeCAD #15892: Can no longer pattern a Subtractive (Eg, MultiTransform a Pocket, etc)

AAllport, 2024-08-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using Mirrored/Linear/MultiTranform tools and selecting a subtractive feature, such as a pocket, is no longer possible.

To reproduce:
- Have a solid
- Make a subtractive feature on that solid
- Pattern the feature to land **within the same solid**

Eg part: A repeating patterns of holes in a case

### Full version info

```shell
0.22.0dev
FreeCAD_weekly-builds-38459-conda-Windows-x86_64-py311
```

### Subproject(s) affected?

PartDesign

### Anything else?

The part:
[image]

Before: 
[image]

After:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-16:
We need the full [about](https://wiki.freecad.org/About) info (not abridged.  
Can you also test on stable version to see if it's a regression?

AAllport, 2024-08-17:
Ahh, apologies... I have updated the original post
I can confirm it was working in 0.21!

gilmiz, 2024-08-17:
Didn't work for me on Version: 0.22.0dev.38459 (Git) either, but it looks like it's working on the latest weekly-build (Version: 0.22.0dev.38467 (Git)).

[image]

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git)
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Danish/Denmark (da_DK)
Installed mods: 
  * Assembly4 0.50.15
  * dodo 1.0.1
  * fasteners 0.5.24
  * FreeCAD-themes 2024.7.24
  * OpenTheme 2024.7.3
  * Render 2024.7.4
  * sheetmetal 0.4.22

closed 2024-08-17 by AAllport