FreeCAD/FreeCAD #15896: Material: SIGABRT after Tab in 3D array editor

chennes, 2024-08-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When editing the first column of a 3D array in the Material editor, pressing "Tab" in the last value causes a SIGABRT (when compiled in debug mode).

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38131 (Git)
Build type: Release
Branch: main
Hash: 9a51e2c6f1ae8b4a1919f6af0f90df6833ecf82f
Python 3.12.4, Qt 6.7.0, Coin 4.0.2, Vtk , OCC 7.8.1
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.4.20
  * fasteners 0.5.2
  * CurvedShapes 1.0.5
  * sheetmetal 0.4.2
  * Curves 0.6.23
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-08-15:
Backtrace:
```
__pthread_kill 0x00000001846b6a60
pthread_kill 0x00000001846eec20
abort 0x00000001845fba30
abort_message 0x00000001846a5d08
demangling_terminate_handler() 0x0000000184695fc0
_objc_terminate() 0x00000001843341e0
std::__terminate(void (*)()) 0x00000001846a50cc
__cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) 0x00000001846a8348
__cxa_throw 0x00000001846a828c
Materials::Material3DArray::validateDepth(int) const MaterialValue.cpp:529
Materials::Material3DArray::setCurrentDepth(int) MaterialValue.cpp:780
MatGui::Array3D::onSelectDepth(const QItemSelection &, const QItemSelection &) Array3D.cpp:215
QtPrivate::FunctorCall::call(void (MatGui::Array3D::*)(const QItemSelection &, const QItemSelection &), MatGui::Array3D *, void **) qobjectdefs_impl.h:145
QtPrivate::FunctionPointer::call<…>(void (MatGui::Array3D::*)(const QItemSelection &, const QItemSelection &), MatGui::Array3D *, void **) qobjectdefs_impl.h:182
QtPrivate::QCallableObject::impl(int, QtPrivate::QSlotObjectBase *, QObject *, void **, bool *) qobjectdefs_impl.h:553
void doActivate<false>(QObject*, int, void**) 0x000000010c309ca8
QItemSelectionModel::emitSelectionChanged(QItemSelection const&, QItemSelection const&) 0x000000010c471cdc
QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>) 0x000000010c471820
QItemSelectionModel::select(QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) 0x000000010c471684
QItemSelectionModel::setCurrentIndex(QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) 0x000000010c472088
QAbstractItemView::keyPressEvent(QKeyEvent*) 0x000000010b74310c
QAbstractItemView::focusNextPrevChild(bool) 0x000000010b73ff08
..
```

luzpaz, 2024-08-22:
Hey @davesrocketshop any availability to check this out?

davesrocketshop, 2024-08-22:
I'm in that code now. I'll look at it today

davesrocketshop, 2024-12-03:
This has not yet been addressed. It will be tackled soon as the many editor issues are addressed together

spanner888, 2026-05-05:
With released versions of FC 1.0.0 and 1.1.1 the same "pressing "Tab" in the last value" triggers what appears to be same issue.

In both below, Open FC, Open Materials WB, Open Materials Editor, Navigate to Test-Test material, in right hand pane, select Physical tab, double click on icon for TestArray3D, then:

For FC 1.0.0: Error occurs if Tab to last cell, or 1st click is on last cell of the 3D index, ie Temperature:
`Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: FreeCAD Exception
Log: The event type 2 was sent to QWidget
Object tree:
	QWidget (qt_scrollarea_viewport) is child of
	QTableView (table3D) is child of
	QSplitter (splitter) is child of
	MatGui::Array3D (MatGui__Array3D)`

For FC 1.1.1: slightly different error message, but same triggers:
`Log: Materials_Edit
Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: FreeCAD Exception
Log: The event type 6 was sent to QTableView
Object tree:
	QTableView (table3D) is child of
	QSplitter (splitter) is child of
	MatGui::Array3D (MatGui__Array3D)Log: Hide main window`

`OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CamScripts`

`OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.1.20260414 (Git shallow) AppImage
Build date: 2026/04/14 22:09:59
Build type: Release
Branch: (HEAD detached at 0108fd4b4)
Hash: 0108fd4b4850cc46e625b60e53cea7a7bbe69f8d
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.4, OCC 7.8.1
Locale: English/Australia (en_AU)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/100.707/1
Installed mods: 
  * Supplemental-Materials 1.0.2
  * CamScripts
  * Woods 1.1.0
  * Nodes 0.1.36
  * Machines 1.0.0`