FreeCAD/FreeCAD #15899: Items are not updating correctly when changing the `Direction/edge:` to Custom on Part Design Pad and Pocket.

leoheck, 2024-08-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Items are not updating correctly when changing the `Direction/edge:` to Custom on Part Design Pad and Pocket.

These are the initial default settings (with the `Sketch normal` selected).
[image]

Changing to `Custom direction`
[image]

Changing back to `Sketch normal` the settings remain the same as in the Custom direction. 
[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36438 (Git) AppImage
Build type: Release
Branch: main
Hash: 0def330a9e2119a14bb349cc406c2aa365421ab4
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.24
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.8.15
  * fasteners 0.5.26
  * asm4_disable_version_check_popup
  * OpticsWorkbench 1.0.20
  * lattice2 1.0.0
  * kicadStepUpMod
  * OpticsWorkbench.backup1722883583.3368473 1.0.18 (Disabled)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-08-19:
I can't reproduce this. Can it be that the sketch normal in your file is also -1,0,0?

leoheck, 2024-12-03:
lol

tarman3, 2024-12-03:
Can not reproduce

https://github.com/user-attachments/assets/0c875b65-513e-4d47-8494-798abf176380

<details>
  <summary>1.1.0dev.39416 (Git) Conda AppImage, Arch Linux (KDE/plasma/wayland)</summary>
```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51
```
</details>

leoheck, 2024-12-03:
I am not talking about the model. I am talking about the Task menu itself. The issue is visible in your video. When you switch back to the default mode. The fields are disabled instead of going back to the default (hidden)