FreeCAD/FreeCAD #15901: [Bug] TechDraw - Radius Dimension Error, Intermitent Functionality

sparker0594, 2024-08-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The issue in the TechDraw module when dimensioning the 0.25mm corner radius on this simple revolved part.
If I dimension the radius from the bottom section of the Front View, the radius dimensions ok.
If I do the same with the top section of the Front View, it will give me a liner-style dimension with a semi-circle symbol before the dimension text, almost like this could be dimensioning the circumference of the corner radius.

If I use the 'TechDraw_RadiusDimension' Tool it will produce the error: "Selected edge is a B-spline and a radius/diameter can not be calculated"
This shouldn't be a B-spline as it's the side view of the 'PartDesign_Fillet' tool.

GIF showing the issue: 
[image]

 GIF of the error message if forcing the radius dimension, instead of using the multi-dimension 'TechDraw_Dimension' tool: 
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods:
* 3DfindIT 1.2.0
* A2plus 0.4.65
* CurvedShapes 1.0.9
* Curves 0.6.42
* sheetmetal 0.4.23
```

### Subproject(s) affected?

TechDraw

### Anything else?

[14D-49226 (Hembrug Chuck Centre) with drawing.zip](https://github.com/user-attachments/files/16636751/14D-49226.Hembrug.Chuck.Centre.with.drawing.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-16:
does this work in the stable version?

WandererFan, 2024-08-17:
That small arc is definitely a spline.  OCC loves splines and uses them for curves that are/should be lines and circles.
We try to interpret these splines as lines/circles.  There were some recent changes in spline.asCircle that might need revisiting.

sparker0594, 2024-08-17:
> does this work in the stable version?

Using 0.21.2 the dimensional are produced as expected:
[image]

sparker0594, 2024-08-17:
> That small arc is definitely a spline. OCC loves splines and uses them for curves that are/should be lines and circles. We try to interpret these splines as lines/circles. There were some recent changes in spline.asCircle that might need revisiting.

Thank you for letting me know I know more about how it works :) (also I had to look up what OCC was, though again, good to know!)

The stable 0.21.2 version produces the expected result.

closed 2024-08-19 by WandererFan