FreeCAD/FreeCAD #15902: CAM: If a toolbit is loaded strange messages are shown in the report view

jkvfc, 2024-08-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If a job is added to the attached workpiece
[image]
[TestDrilling.zip](https://github.com/user-attachments/files/16637176/TestDrilling.zip)
the following messages are shown in the report view
[image]
The messages are obviously engendered if a toolbit is loaded, because loading for instance the 5 mm drill of FreeCAD's default library raises the messages
[image]
Such messages does not appear in version 0.21.2.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38462 (Git)
Build type: Unknown
Branch: main
Hash: 0d067978cdb6bdfa8c2c2aad60eae33a3369532c
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: English/United States (en_US)
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-17:
was the file created in 0.21? What happens when you recompute the document, save with 0.22 and then try again?

jkvfc, 2024-08-17:
The file was created with the above mentioned version 0.22.0dev.38462 (Git).
Recomputing the document and saving it under a new name and then loading a new toolbit causes the same messages.
Doing the same operations in 0.21.2 never yields such messages.

maxwxyz, 2024-08-17:
@spanner888 @sliptonic any idea about this issue?

spanner888, 2024-08-17:
I can confirm in recent weekly's.

Issue seems to be the default shape files need to be opened, fully recomputed, including the sketch and saved so the recent Topo work is applied. Any FC backup files need to be removed.

Had been aware of this for a while and have been extensively testing this via unrelated work I am doing. Was not rushing fix, in case more topo changes impacted again.

Reconfirming on endmill.fcstd just now triggered an error which I had not seen before, but will look into:

`Reading failed from embedded file: StringHasher.Table.txt (22 bytes, 24 bytes compressed)`

sliptonic, 2024-08-17:
I went ahead and resaved them.  Makes me wonder if a little utilty script should be written to open, recompute, and resave these files.  Not necessarily as part of CI/CD process but just to save some grunt work.

closed 2024-08-19 by WandererFan

spanner888, 2024-08-20:
So I tried @sliptonic 's challenge to create a script. It was little, until I tried to work around FreeCAD change internal names of files with "-".

At present it runs and saves, but cannot close the two files that have "-" in name, because of this FC issue. Only tested on Debian.

Did create #15966 with details.

So even to me this code is starting to look unreliable to mass promote to users. Any comments, or should I ask the forum?

`f_extension = ".fcstd"
sys_tool = "endmill" + f_extension

def is_doc_name_changed(fname, silent=True, msg=""):
	name_now = FreeCAD.ActiveDocument.Name
	if name_now == fname:
		return False
	else:
		if not silent:
			print("{}Name changed now from {} to {}.".format(msg, f, name_now))
		return True

sys_shape_dir = ""
em=Path.Tool.Bit.findToolShape(sys_tool)
if len(em) > len(sys_tool):
	if sys_tool and em.endswith(sys_tool):
		sys_shape_dir = em[:-len(sys_tool)]

print(sys_shape_dir)
for f in os.listdir(sys_shape_dir):
	if f.endswith(f_extension):
		FreeCAD.openDocument(sys_shape_dir + f)

		f = f[:-len(f_extension)]
		name_as_opened = FreeCAD.ActiveDocument.Name
		print("requested open {}, opened name {}.".format(f, name_as_opened))
		is_doc_name_changed(f, msg="Opened ")
		FreeCAD.activeDocument().recompute(None,True,True)
		# is_doc_name_changed(f, msg="recompute ")

		if is_doc_name_changed(f, silent=True, msg="before save "):
			print("\tDocument name changed from {}, to {}.".format(f, name_as_opened))
			full_name = sys_shape_dir + f + f_extension
			FreeCAD.ActiveDocument.saveAs(full_name)
			print("\t\tDoc *savedAS* {}: {}".format(f, full_name))

			# Can't close document:' <class 'NameError'>: Unknown document 'thread-mill'
			# FreeCAD.closeDocument(f)
		else:
			FreeCAD.getDocument(f).save()
			FreeCAD.closeDocument(f)`

commit 6a23cd01 (2024-08-19): fixes #15902