FreeCAD/FreeCAD #15904: Crash when closing the "Transform" task

wiz78, 2024-08-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With this weekly build, whenever you transform an object, it crashes when you close the task.

To reproduce it:

- create a new document
- create a new body
- open the contextual menu on the new body and choose "Transform"
- either confirm or cancel the task

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.87.0
  * ProDarkThemePreferencePack 1.0.0
  * freecad.gears 1.0.0
  * Manipulator 1.5.0
  * fasteners 0.4.56
```

### Subproject(s) affected?

PartDesign

### Anything else?

[crashlog.txt](https://github.com/user-attachments/files/16638427/crashlog.txt)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-08-16:
No crash here. Looks like a bug in _Coin_.
```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38465 (Git)
Build type: Debug
Branch: main
Hash: dc6b7ce3ed7350118e4ff7e48f56ccbd760c3dc5
Python 3.10.12, Qt 5.15.14, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

```ruby
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
```

wiz78, 2024-08-16:
It might be macOS-related and it might be a return of https://github.com/FreeCAD/FreeCAD/issues/9465

maxwxyz, 2024-08-17:
@johnocfii could you confirm this?

JohnOCFII, 2024-08-17:
> @JohnOCFII could you confirm this?

It crashed for me, but only if you acted on the `Transform` (which, of course, makes sense, but based on the steps included above, first time I tried to replicate it, I selected `Transform` and then quit out of the Transform without actually grabbing a handle and moving the object).

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [1752]
Path:                  /Applications/Other Apps/FreeCAD-38459.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [1740]
User ID:               501

Date/Time:             2024-08-17 10:17:02.6025 -0500
OS Version:            macOS 14.6.1 (23G93)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       5411344A-DD80-42BD-B0E7-26696599A4D5

Time Awake Since Boot: 390000 seconds
Time Since Wake:       1245 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Codes:       0x0000000000000001, 0x0000000000000010

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [1752]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 4306288624
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100acc000-100aec000    [  128K] r-x/r-x SM=COW  /Applications/Other Apps/FreeCAD-38459.app/Contents/Resources/bin/freecad

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libCoin.80.0.2.dylib          	       0x1053995a4 SoCallbackAction::~SoCallbackAction() + 124
1   libCoin.80.0.2.dylib          	       0x1053996d4 SoCallbackAction::~SoCallbackAction() + 12
2   libCoin.80.0.2.dylib          	       0x1053f6ea0 SoDragger::~SoDragger() + 192
3   libFreeCADGui.dylib           	       0x102854b6c Gui::RDragger::~RDragger() + 12
4   libCoin.80.0.2.dylib          	       0x1055375a8 SoBase::destroy() + 212
5   libCoin.80.0.2.dylib          	       0x10552b674 SoBaseList::truncate(int) + 80
6   libCoin.80.0.2.dylib          	       0x10553b8c4 SoChildList::~SoChildList() + 24
7   libCoin.80.0.2.dylib          	       0x105587318 SoGroup::~SoGroup() + 40
8   libCoin.80.0.2.dylib          	       0x1055a1600 SoSeparator::~SoSeparator() + 12
9   libCoin.80.0.2.dylib          	       0x1055375a8 SoBase::destroy() + 212
10  libCoin.80.0.2.dylib          	       0x10552b674 SoBaseList::truncate(int) + 80
11  libCoin.80.0.2.dylib          	       0x10553b8c4 SoChildList::~SoChildList() + 24
12  libCoin.80.0.2.dylib          	       0x105587318 SoGroup::~SoGroup() + 40
13  libCoin.80.0.2.dylib          	       0x1055a53d0 SoSwitch::~SoSwitch() + 56
14  libCoin.80.0.2.dylib          	       0x1055375a8 SoBase::destroy() + 212
15  libCoin.80.0.2.dylib          	       0x10552b674 SoBaseList::truncate(int) + 80
16  libCoin.80.0.2.dylib          	       0x10553b8c4 SoChildList::~SoChildList() + 24
17  libCoin.80.0.2.dylib          	       0x105587318 SoGroup::~SoGroup() + 40
18  libFreeCADGui.dylib           	       0x102843e7c Gui::So3DAnnotation::~So3DAnnotation() + 12
19  libCoin.80.0.2.dylib          	       0x1055375a8 SoBase::destroy() + 212
20  libCoin.80.0.2.dylib          	       0x10552b674 SoBaseList::truncate(int) + 80
21  libCoin.80.0.2.dylib          	       0x10553b8c4 SoChildList::~SoChildList() + 24
22  libCoin.80.0.2.dylib          	       0x105587318 SoGroup::~SoGroup() + 40
23  libCoin.80.0.2.dylib          	       0x1055a1600 SoSeparator::~SoSeparator() + 12
24  libCoin.80.0.2.dylib          	       0x1055375a8 SoBase::destroy() + 212
[... 20 lines cut ...]
45  CoreFoundation                	       0x19dddf434 CFRunLoopRunSpecific + 608
46  HIToolbox                     	       0x1a858919c RunCurrentEventLoopInMode + 292
47  HIToolbox                     	       0x1a8588e2c ReceiveNextEventCommon + 220
48  HIToolbox                     	       0x1a8588d30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
49  AppKit                        	       0x1a163ecc8 _DPSNextEvent + 660
50  AppKit                        	       0x1a1e354d0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
51  AppKit                        	       0x1a1631ffc -[NSApplication run] + 476
52  libqcocoa.dylib               	       0x1066944d0 0x106668000 + 181456
53  freecad                       	       0x100ad3204 main + 3356
54  dyld                          	       0x19d977154 start + 2476
```

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.8.10
```

closed 2024-08-18 by chennes

theosib, 2024-08-25:
I think I'm having the same problem in 38459. In the document attached, I select the Cone, choose Transform, and use the red circle to rotate the cone 90 degrees around the X axis. When I click OK, FreeCAD crashes.
[book binding.FCStd.zip](https://github.com/user-attachments/files/16741674/book.binding.FCStd.zip)

maxwxyz, 2024-08-25:
@theosib please use the newest weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

PR #15924 (2024-08-18): MacOS: Modify OS detection macro
Attempt to fix #15904 -- maybe on the Conda build system the macros are defined differently?

ETA: tested with a local Conda build and this does solve the transform problem.
commit ff1409d8 (2024-08-18): MacOS: Modify OS detection macro (#15924)

* Fixes #15904
* MacOS: Always use Q_OS_MACOS to detect MacOS because the Conda builds do not define Q_OS_APPLE