FreeCAD/FreeCAD #15905: Nightly -- 5 things

FlailAway, 2024-08-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1: Saving to .step (etc) adds another .step to the filename - e.g : plug.step.step.step

2: Fails to redraw "As-Is" after wire-frame, stays as wire-frame until I close and restart

3: The "don't show at start" checkbox only showed up after 3 restarts (fluff -- I know)

4: Often displays limited set of tools and have to close and reopen Sketch to get them back. (pic attached) I want "Tangent" but it ain't there.

5: "Symetric" shows orange error lines trying to constrain a square horizontally about center-line as the first item in a new Sketch.

### Full version info

```shell
OS: Linux Mint 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git) AppImage
Build type: Release
Branch: main
Hash: 58729b3ab0060aeabfd4b3d5e22e1525fc45c87b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Anything else?

[freecad-edit-tools-01.jpg.zip](https://github.com/user-attachments/files/16639651/freecad-edit-tools-01.jpg.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-16:
We need to split this ticket.

depthoffocus, 2024-08-17:
Number 4 is definitely #13243.

depthoffocus, 2024-08-17:
And 2. is, I think, an issue I have raised myself: #15352

maxwxyz, 2024-08-17:
I cannot reproduce 1-3 and 4 is already a duplicate. Please open separate issues for each one, ideally with an example file and steps how to reproduce.

closed 2024-08-17 by maxwxyz

FlailAway, 2024-09-07:
@maxwxyz:
1: Are you using Linux? I use "Ctr-E" to open the save dialog for a second time, same session and my default is set to ".step." It shows the filename and I click "OK" and it saves it as "whatever.step.step" If I make changes to the project again, same session and "Ctrl-E" again it gives "whatever.step.step.step" 

3: my fault for not clicking the "Done" or something button way over on the right side for the first time use of FC. As a retired programmer for roughly 50-years, I'd be happy to help with advice on HID (Human Interface Design) ;)

luzpaz, 2024-09-07:
@FlailAway we'd be happy to hear more of your feedback. It's easier for us if you open separate issues for each thing. TIA