FreeCAD/FreeCAD #15906: Joints become locked when s

PCS-Rimage, 2024-08-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

To learn about the new integrated assembly WB I made a small project with a lever arm using the cyclindrical joint.  It worked great when I made the project as I was able to move the lever in theta and axially.  After closing and reopening the project I cannot move the lever in either dimension and cannot figure out how to get it moving again.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git)
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * Assembly4 0.50.15
  * dodo 1.0.1
  * freecad.gears 1.2.0
  * Help 1.0.3
  * lattice2 1.0.0
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

None

### Anything else?

[AssyTest.zip](https://github.com/user-attachments/files/16643300/AssyTest.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-17:
Double click the assembly object.
Please use the wiki or the forum for help request in the future.

closed 2024-08-17 by maxwxyz

PCS-Rimage, 2024-08-17:
Thanks for the tip.  I was setting the assembly Active object flag via right click instead of double-click. I guess they are diffent even though double-click toggles the flag state.