FreeCAD/FreeCAD #15909: CNC file import pre-processors not working + unclear UI

alexfacciorusso, 2024-08-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The GCode import function introduced by https://tracker.freecad.org/view.php?id=3455 is not working. The GCode is not pre-processed properly and an error is displaying when trying to select GCode preprocessor from the dialog. 

In addition, the dialog is not very clear to the final user and contains "example" preprocessors in the release folder.

### Full version info

```shell
When trying to import/open GCode and pre-process it via one of the pre-defined pre-processors, an error is 
displayed.
```

### Subproject(s) affected?

CAM

### Anything else?

https://tracker.freecad.org/view.php?id=3455

### Step 1 - open the file
[image]

### Step 2 - dialog displayed

You can see the dialog having code-friendly names instead of user-friendly ones. Also, no description on what each script does.

[image]

### Step 3 - select gcode_pre

[image]

### Step 4 - Code not imported and error displayed

[image]

```
11:11:14  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\freecad\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'gcode_pre'
11:11:14  pyException: Traceback (most recent call last):
  File "<string>", line 5, in <module>
<class 'RuntimeError'>: No module named 'gcode_pre'
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-17:
Need full [About](https://wiki.freecad.org/About) info please

alexfacciorusso, 2024-08-17:
> Need full [About](https://wiki.freecad.org/About) info please

Added to the main bug data

PR #15921 (2024-08-17): [CAM] Fix Gcode import 
Fixes #15909  
regression bug caused by Path->CAM module rename.
commit 8dc1fad1 (2024-08-17): Fixes #15909
regression bug caused by Path->CAM module rename.
closed 2024-08-17 by sliptonic