FreeCAD/FreeCAD #15911: Incomplete functionality of new measure tool

macdroid53, 2024-08-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The new measure tool functionality is limited. A major application of such a tool, especially in mechanical designs, is to measure the the center to center distance of circles/shafts, or center to edge, etc. Also, any measurement tool should allow the selection of intersecting and vertexes as the user chooses. There appears to to be no way to accomplish with these in current implementation. 

### Full version info

```shell
0.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-17:
@macdroid53 please open a more succinct ticket that specifies 1 specific feature. There is a measurment tag that you can reference: https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aopen+is%3Aissue+label%3A%22Topic%3A+Measurement%22

closed 2024-08-17 by luzpaz

macdroid53, 2024-08-17:
I was trying to be gentle. The tool should not be offered in it's current state. Several have already suggested the old one should not have been removed until the new has more fleshed out features.
Currently, you still need other tools to have a complete measurement capability.

luzpaz, 2024-08-17:
@macdroid53 can you interface directly with the devs involved on the forum. This decision was made some time ago. i believe it was wandererfan, hlorus and others.