FreeCAD/FreeCAD #15917: Measurement: user selectable snapping

macdroid53, 2024-08-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The tool needs to have user selectable snapping for center points, midpoint, end vertex, etc.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2024-08-17:
To add some detail to this issue: The measurement utility lacks the ability to snap measurements to key points between certain geometry elements. A key problem is the ability to measure center-to-center between circles/holes. An example of a viable approach to managing this is demonstrated by the "manipulator" workbench addon.

macdroid53, 2024-08-17:
As well as the Manipulator workbench Caliper tool, other examples exist in Assembly4 and Quick Measure.

maxwxyz, 2024-08-17:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/13708

closed 2024-08-17 by maxwxyz

macdroid53, 2024-08-17:
It was said that the lacking functionality of this tool had not been reported.

As noted above, it has been, in #13708  and in that issue there is a link to the forum thread where these were reported.