FreeCAD/FreeCAD #15918: Segfault on sketch closing

nazar-pc, 2024-08-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I open one document that has another document linked in it, open sketch, then switch to a spreadsheet from liked document and try to close sketch that was still open and FreeCAD crashed:

<details>

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x77ccbb645320]
#1  0x77ccbb1bd781 in SoRenderManager::setRenderMode(SoRenderManager::RenderMode) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/./libCoin.so.80+0x1
#2  0x77ca32358915 in SketcherGui::ViewProviderSketch::unsetEdit(int) from /tmp/.mount_FreeCAm7srhc/usr/lib/SketcherGui.so+0x155
#3  0x77ccbed5ed43 in Gui::ViewProvider::finishEditing() from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libFreeCADGui.so+0x13
#4  0x77ccbe8fd6fd in Gui::Document::_resetEdit() from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libFreeCADGui.so+0x9d
#5  0x77ccbe87cada in Gui::Application::setEditDocument(Gui::Document*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libFreeCADGui.so+0x3a
#6  0x77ccbe9121b0 in Gui::DocumentPy::resetEdit(_object*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libFreeCADGui.so+0x30
#7  0x77ccbe9121e4 in Gui::DocumentPy::staticCallback_resetEdit(_object*, _object*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libFreeCADGui.so+0x24
#8  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x77ccbe006068]
#9  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x77ccbdfe5b5c]
#10  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x77ccbdff1532]
#11  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x77ccbe0abd1d]
#12  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x77ccbe0ab794]
#13  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x77ccbe0cabf3]
#14  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x77ccbe0c669a]
#15  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x77ccbe0ba9c9]
#16  0x77ccbd33d9fc in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libFreeCADBase.so+0x6c
#17  0x77ccbe989b24 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libFreeCADGui.so+0x94
#18  0x77ccbe989cb3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libFreeCADGui.so+0xe3
#19  0x77ca32394725 in SketcherGui::TaskDlgEditSketch::reject() from /tmp/.mount_FreeCAm7srhc/usr/lib/SketcherGui.so+0x145
#20  0x77ccbecc9bb7 in Gui::TaskView::TaskView::reject() from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libFreeCADGui.so+0x67
#21  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x77ccbbceb6f2]
#22  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Widgets.so.5(+0x2fab70) [0x77ccbccfab70]
#23  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x77ccbbceb964]
#24  0x77ccbcc5b522 in QAbstractButton::clicked(bool) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Widgets.so.5+0x32
#25  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Widgets.so.5(+0x25b73a) [0x77ccbcc5b73a]
#26  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Widgets.so.5(+0x25d1c8) [0x77ccbcc5d1c8]
#27  0x77ccbcc5d3d7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Widgets.so.5+0xe7
#28  0x77ccbcbaec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Widgets.so.5+0x178
#29  0x77ccbcb703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#30  0x77ccbcb78262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#31  0x77ccbe93e419 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libFreeCADGui.so+0x99
#32  0x77ccbbcb6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Core.so.5+0x108
#33  0x77ccbcb763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#34  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x77ccbcbc6e6f]
#35  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x77ccbcbca0c9]
#36  0x77ccbcb703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#37  0x77ccbe93e419 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libFreeCADGui.so+0x99
#38  0x77ccbbcb6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Core.so.5+0x108
#39  0x77ccbc3353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Gui.so.5+0x84c
#40  0x77ccbc30b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Gui.so.5+0xac
#41  /tmp/.mount_FreeCAm7srhc/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x77ccb6494bea]
#42  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x77ccba310ad7]
#43  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x77ccba313ac7]
#44  /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x77ccba3140ce]
#45  0x77ccbbd0aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Core.so.5+0x66
#46  0x77ccbbcb570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Core.so.5+0x11b
#47  0x77ccbbcbd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libQt5Core.so.5+0x82
#48  0x77ccbe89f8e3 in Gui::Application::runApplication() from /tmp/.mount_FreeCAm7srhc/usr/bin/../lib/libFreeCADGui.so+0x983
#49  /tmp/.mount_FreeCAm7srhc/usr/bin/freecad(+0x7620) [0x568de8194620]
#50  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x77ccbb62a1ca]
#51  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x77ccbb62a28b]
#52  /tmp/.mount_FreeCAm7srhc/usr/bin/freecad(+0x7981) [0x568de8194981]
```

</details>

Demo:

https://github.com/user-attachments/assets/c677f368-d0d6-4dac-8f02-b508ecd30479

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git) AppImage
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * parts_library
  * Curves 0.6.42
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-08-17:
The crash is easily reproducible on both _Linux_ and _Windows_.

<details><summary>Ubuntu 22.04.4 LTS (KDE/plasma)</summary>

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38465 (Git)
Build type: Debug
Branch: main
Hash: dc6b7ce3ed7350118e4ff7e48f56ccbd760c3dc5
Python 3.10.12, Qt 5.15.14, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

```cpp
(lldb) bt
error: libFreeCADGui.so 0x07b12ce8: DW_TAG_member '_M_local_buf' refers to type 0x0000000007b66fa2 which extends beyond the bounds of 0x07b12cde
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x8)
  * frame #0: 0x00007ffff2cd23a5 libCoin.so.80c`SoRenderManager::setRenderMode(SoRenderManager::RenderMode) + 5
    frame #1: 0x00007ffff7b9fd30 libFreeCADGui.so`Gui::View3DInventorViewer::updateOverrideMode(this=0x00005555595bd6c0, mode="As Is") at View3DInventorViewer.cpp:1169:50
    frame #2: 0x00007fffa08eaf48 SketcherGui.so`SketcherGui::ViewProviderSketch::unsetEdit(this=0x000055555960d0e0, ModNum=<unavailable>) at ViewProviderSketch.cpp:3330:43
    frame #3: 0x00007ffff7bdcbef libFreeCADGui.so`Gui::ViewProvider::finishEditing(this=0x000055555960d0e0) at ViewProvider.cpp:153:14
    frame #4: 0x00007ffff780b4fe libFreeCADGui.so`Gui::Document::_resetEdit(this=0x0000555555e47f30) at Document.cpp:481:44
    frame #5: 0x00007ffff77b02c6 libFreeCADGui.so`Gui::Application::setEditDocument(this=0x00007fffffffd7f0, doc=0x0000000000000000) at Application.cpp:1153:29
    frame #6: 0x00007ffff7806724 libFreeCADGui.so`Gui::Document::resetEdit(this=<unavailable>) at Document.cpp:468:43
    frame #7: 0x00007ffff783c78c libFreeCADGui.so`Gui::DocumentPy::resetEdit(this=0x000055555b916310, args=<unavailable>) at DocumentPyImp.cpp:169:32
    frame #8: 0x00007ffff783c7d5 libFreeCADGui.so`Gui::DocumentPy::staticCallback_resetEdit(self=0x000055555b916318, args=<unavailable>) at DocumentPy.cpp:643:66
    frame #9: 0x00007ffff5128008 libpython3.10.so.1.0`___lldb_unnamed_symbol3713 + 72
    frame #10: 0x00007ffff50dfadc libpython3.10.so.1.0`_PyObject_MakeTpCall + 140
    frame #11: 0x00007ffff507ba1c libpython3.10.so.1.0`_PyEval_EvalFrameDefault + 40380
    frame #12: 0x00007ffff51c23af libpython3.10.so.1.0`___lldb_unnamed_symbol4418 + 1647
    frame #13: 0x00007ffff51bd3de libpython3.10.so.1.0`PyEval_EvalCode + 190
    frame #14: 0x00007ffff520c4cd libpython3.10.so.1.0`___lldb_unnamed_symbol4668 + 157
    frame #15: 0x00007ffff520d4f9 libpython3.10.so.1.0`PyRun_StringFlags + 121
    frame #16: 0x00007ffff5801715 libFreeCADBase.so`Base::InterpreterSingleton::runString[abi:cxx11](this=<unavailable>, sCmd="Gui.getDocument('Unnamed').resetEdit()") at Interpreter.cpp:249:15
    frame #17: 0x00007ffff789ad0c libFreeCADGui.so`Gui::Command::_runCommand(file="/home/xtemp09/Downloads/FreeCAD/freecad-source/src/Mod/Sketcher/Gui/TaskDlgEditSketch.cpp", line=137, eType=Gui, sCmd="Gui.getDocument('Unnamed').resetEdit()") at Command.cpp:725:38
    frame #18: 0x00007ffff789aeca libFreeCADGui.so`Gui::Command::_doCommand(file="/home/xtemp09/Downloads/FreeCAD/freecad-source/src/Mod/Sketcher/Gui/TaskDlgEditSketch.cpp", line=137, eType=Gui, sCmd="") at Command.cpp:677:16
    frame #19: 0x00007fffa092ba9f SketcherGui.so`SketcherGui::TaskDlgEditSketch::reject(this=0x0000555559f49920) at TaskDlgEditSketch.cpp:137:19
    frame #20: 0x00007ffff7b4eac8 libFreeCADGui.so`Gui::TaskView::TaskView::reject(this=0x0000555555c360f0) at TaskView.cpp:827:40
    frame #21: 0x00007ffff7b4fe92 libFreeCADGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::TaskView::TaskView::*)()>::call(f=<unavailable>, o=<unavailable>, arg=<unavailable>) at qobjectdefs_impl.h:152:20
    frame #22: 0x00007ffff7b4fed0 libFreeCADGui.so`QtPrivate::QSlotObject<void (Gui::TaskView::TaskView::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) [inlined] void QtPrivate::FunctionPointer<void (Gui::TaskView::TaskView::*)()>::call<QtPrivate::List<>, void>(arg=<unavailable>, o=<unavailable>, f=<unavailable>) at qobjectdefs_impl.h:185:95
    frame #23: 0x00007ffff7b4fec8 libFreeCADGui.so`QtPrivate::QSlotObject<void (Gui::TaskView::TaskView::*)(), QtPrivate::List<>, void>::impl(which=<unavailable>, this_=<unavailable>, r=<unavailable>, a=<unavailable>, ret=<unavailable>) at qobjectdefs_impl.h:418:49
    frame #24: 0x00007ffff3cf819f libQt5Core.so.5`void doActivate<false>(QObject*, int, void**) [inlined] QtPrivate::QSlotObjectBase::call(a=0x00007fffffffbe90, r=0x0000555555c360f0, this=0x000055555947ea80) at qobjectdefs_impl.h:398:57
    frame #25: 0x00007ffff3cf8183 libQt5Core.so.5`void doActivate<false>(sender=0x0000555556c5cb40, signal_index=10, argv=0x00007fffffffbe90) at qobject.cpp:3925:30
    frame #26: 0x00007ffff3cef16f libQt5Core.so.5`QMetaObject::activate(sender=<unavailable>, m=0x0000000000737280, local_signal_index=3, argv=0x0000000000000000) at qobject.cpp:3985:26
    frame #27: 0x00007ffff4b48984 libQt5Widgets.so.5`QDialogButtonBox::rejected(this=<unavailable>) at moc_qdialogbuttonbox.cpp:299:26
    frame #28: 0x00007ffff4b48ff6 libQt5Widgets.so.5`QDialogButtonBoxPrivate::_q_handleButtonClicked(this=<unavailable>) at qdialogbuttonbox.cpp:882:29
[... cut ...]
#48 0x00007ffa0345421f in FreeCADGui!?runApplication@Application@Gui@@SAXXZ ()
   from E:\temp\FreeCAD_weekly-builds-38459-conda-Windows-x86_64-py311\bin\FreeCADGui.dll
#49 0x00007ff79fab152e in ?? ()
#50 0x00007ff79fab3d47 in ?? ()
#51 0x00007ff79fab1ec6 in ?? ()
#52 0x00007ffa66487374 in KERNEL32!BaseThreadInitThunk () from C:\Windows\System32\kernel32.dll
#53 0x00007ffa66edcc91 in ntdll!RtlUserThreadStart () from C:\Windows\SYSTEM32\ntdll.dll
#54 0x0000000000000000 in ?? ()
(gdb)

```

</details>

maxwxyz, 2024-08-17:
Is this a regression or already happening in the stable 0.21.2 version?

xtemp09, 2024-08-17:
No crash here.

```ruby
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

xtemp09, 2024-08-19:
@maxwxyz, the commit maxwxyz@4e8f3f03817d433692a823d981cb5ab96c0290d3 seems to introduce the segmentation fault. The code thinks that _Sketcher_ is opened.

maxwxyz, 2024-08-19:
thanks, unfortunately I'm not able to provide a fix

AjinkyaDahale, 2024-08-21:
> the commit [maxwxyz/FreeCAD@4e8f3f0](https://github.com/maxwxyz/FreeCAD/commit/4e8f3f03817d433692a823d981cb5ab96c0290d3) seems to introduce the segmentation fault. Within the code snippet _Sketcher_ is presumed to be active.

There are a bunch of dereferences added there. Is it guaranteed that `hGrp` can never be null?

maxwxyz, 2024-09-01:
related: https://github.com/FreeCAD/FreeCAD/issues/15352

AjinkyaDahale, 2024-09-01:
@maxwxyz, would you be able to fix this on basis of [comments](https://github.com/maxwxyz/FreeCAD/commit/4e8f3f03817d433692a823d981cb5ab96c0290d3#r145615884) on 4e8f3f03817d433692a823d981cb5ab96c0290d3?

maxwxyz, 2024-09-01:
@AjinkyaDahale I'm not sure how to identify if the viewer is a sketcher viewer, any idea?

AjinkyaDahale, 2024-09-01:
> @AjinkyaDahale I'm not sure how to identify if the viewer is a sketcher viewer, any idea?

Does it not have a `getClassTypeId` or `getTypeId`?

benj5378, 2024-09-01:
I don't think the issue is here. No null pointers either. The seg fault happens when `overrideMode = mode` even though both are std::string. Hmmm...

What is weird though is that
```cpp
Gui::View3DInventor* inventor = static_cast<Gui::View3DInventor*>(mdi);  // Added extra line
Gui::View3DInventorViewer* viewer = inventor->getViewer();

(gdb) p inventor
$5 = (Gui::View3DInventor *) 0x7ffff7e6f0a8 <typeinfo for Gui::View3DInventor>
(gdb) p viewer
$4 = (Gui::View3DInventorViewer *) 0x7fffb034e6f0 <vtable for TechDrawGui::MDIViewPage+816>
```
the viewer pointer points to a vtable for TechDrawGui??

benj5378, 2024-09-01:
After extensive testing, I figured out what is going on.

https://github.com/FreeCAD/FreeCAD/blob/2a9cf9e8e6e62086948eb15e24d1e69a54f53d97/src/Mod/Sketcher/Gui/ViewProviderSketch.cpp#L3321

`getActiveView`. If I go to TechDraw or Spreadsheet, the active view isn't of the `mdi` for `View3DInventor`. No, it's the mdi for TechDrawGui. Or the mdi for Spreadsheet.

This is confirmed by
```
(gdb) p *mdi
$3 = {<QMainWindow> = {<No data fields>}, <Gui::BaseView> = {<Base::BaseClass> = {_vptr.BaseClass = 0x7fffa110d108 <vtable for SpreadsheetGui::SheetView+696>, [...]
                                                                                                                               ^^^^^^^^^^^^^^^^^^^^^^^^^

(gdb) p *mdi
$1 = {<QMainWindow> = {<No data fields>}, <Gui::BaseView> = {<Base::BaseClass> = {_vptr.BaseClass = 0x7fffb034e678 <vtable for TechDrawGui::MDIViewPage+696>, [...]
                                                                                                                               ^^^^^^^^^^^^^^^^^^^^^^^^
```

Syres916, 2024-09-04:
@benj5378 is [this](https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Fix_Sketcher_Segfault_On_Exit_main) a fix?

benj5378, 2024-09-05:
@Syres916 No, it's a diagnosis. I tried fixing it using some other getview function, it didn't seem like the function worked.

CalligaroV, 2024-09-08:
Hi @nazar-pc!

I may have a fix for this issue.
I opened a PR with a modification that looks fixing it, at least in my local tests.
It PR #16378

Can you please try if this works for you too?

I think I'm missing something, this fix looks too simple to me but it takes into account what @benj5378 found.
Therefor feedback are very welcome from anybody.

TIA!

maxwxyz, 2024-09-16:
@xtemp09 @CalligaroV is this related: 
- https://github.com/FreeCAD/FreeCAD/issues/16570

and also fixed by #16378 ?

CalligaroV, 2024-09-16:
Hi @maxwxyz!

I tried to reproduce that issue with success.
IMO the two are unrelated (this one is caused by not having a 3D view while for the second one the 3D view is used all the time) and the PR for this one doesn't look to solve that issue too, at least in my local tests.

I may have a couple of ideas about what's causing that one.
I'll add my comments about that in the related conversation to keep the topics separated.

In case I'm wrong I'll be happy to continue with this conversation.

Thanks!

pinkavaj, 2024-09-17:
I'm experiencing a crash (not reported yet, sorry), which looks pretty similar but the patch #16378 does not fix it. GDB and Valgrind points to some already deleted MDI (window). The reproduction is a bit non-reliable, but includes switching between various windows (eg Propery View) and then closing the "Start" page when it is active.

I'm mentioning this to raise a question if similar issue can be present in other part of the code too.

(I will create separate issue one I finish investigation, just mentioning now and here to raise avareness.)

CalligaroV, 2024-09-17:
@pinkavaj, sometimes it happens to me too to have that kind of crash with the Start page involved.
However I haven't found yet a way to reproduce it reliably (and never tried much to do so TBH)

If you can find a way to reproduce it reliably then please open a new issue, if we're lucky enough the fix may be easy.

Thank you!

pinkavaj, 2024-09-17:
The (possibly) related issue https://github.com/FreeCAD/FreeCAD/issues/16616 ?

closed 2024-09-20 by chennes