FreeCAD/FreeCAD #15926: Permission denied when doing boolean operation on meshes

ruffson, 2024-08-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I try to do any boolean operation on an imported stl mesh and a cylinder I meshed from solid geometry.
When I try to perfom any such operation I am getting the error "Unknown error occurred while running OpenSCAD.".

The console error, I get is Permission denied. I already had given the FreeCAD flatpak app full system disk permission:

```
╰─❯ flatpak run org.freecadweb.FreeCAD
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Gtk-Message: 12:45:43.380: Failed to load module "canberra-gtk-module"
Gtk-Message: 12:45:43.380: Failed to load module "pk-gtk-module"
Gtk-Message: 12:45:43.380: Failed to load module "canberra-gtk-module"
Gtk-Message: 12:45:43.380: Failed to load module "pk-gtk-module"
(qt.qpa.qgnomeplatform) Could not find color scheme  ""
Qt: Session management error: Could not open network socket
The mesh data structure has some defects
Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "/app/freecad/Mod/OpenSCAD/OpenSCADUtils.py", line 529, in meshoptempfile
    result = callopenscadmeshstring('%s(){%s}' % (opname,meshimports))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADUtils.py", line 493, in callopenscadmeshstring
    tmpfilename = callopenscadstring(scadstr, 'stl')
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADUtils.py", line 238, in callopenscadstring
    outputfilename = callopenscad(inputfilename, outputext=outputext,\
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADUtils.py", line 169, in callopenscad
    return call_openscad_with_pipes(inputfilename, outputfilename, outputext, keepname)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/freecad/Mod/OpenSCAD/OpenSCADUtils.py", line 206, in call_openscad_with_pipes
    p = subprocess.Popen([openscad_executable,"--export-format","csg", "-o", "-", "-"],
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.11/subprocess.py", line 1955, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
<class 'PermissionError'>: [Errno 13] Permission denied: ''
```

Not really sure what is going on. Any help and workarounds are greatly appreciated!

### Full version info

```shell
I am on Fedora 40 and am using the flatpak version of FreeCAD 0.21.2. Fuull About info:

[code]
OS: KDE Flatpak runtime (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Mesh

### Anything else?

It probably does not matter but the mesh I am trying to boolean with is this higiberry case stl from: https://www.printables.com/model/24367-hifiberry-case/files

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-17:
Please try to reproduce on dev version (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) and report back, TIA

ruffson, 2024-08-17:
I tried on the weekly beta dev version and get an error that suggests I don't have openscad executable (but I do have it):

```
╰─❯ ./FreeCAD_weekly-builds-38467-conda-Linux-x86_64-py311.AppImage
FreeCAD 0.22.0, Libs: 0.22.0devR38467 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Hifiberry_case_test#Cylinder
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Hifiberry_case_test#Sphere
The mesh data structure has some defects
Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "/tmp/.mount_FreeCA3Tbrri/usr/Mod/OpenSCAD/OpenSCADUtils.py", line 529, in meshoptempfile
    result = callopenscadmeshstring('%s(){%s}' % (opname,meshimports))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCA3Tbrri/usr/Mod/OpenSCAD/OpenSCADUtils.py", line 493, in callopenscadmeshstring
    tmpfilename = callopenscadstring(scadstr, 'stl')
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCA3Tbrri/usr/Mod/OpenSCAD/OpenSCADUtils.py", line 238, in callopenscadstring
    outputfilename = callopenscad(inputfilename, outputext=outputext,\
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCA3Tbrri/usr/Mod/OpenSCAD/OpenSCADUtils.py", line 186, in callopenscad
    raise OpenSCADError('OpenSCAD executable unavailable')
<class 'OpenSCADUtils.OpenSCADError'>: 'OpenSCAD executable unavailable'
```

Openscad:

```
╰─❯ which openscad
/usr/bin/openscad
```

ruffson, 2024-08-17:
Fixed in on the weekly build by specifying a transfer directory that is writeable. For the release, this does not work (and the error is also slightly different)

luzpaz, 2024-08-18:
Can you elaborate the workaround you did, for posterity ?

ruffson, 2024-08-18:
> Can you elaborate the workaround you did, for posterity ?

So when I was testing this and because I started the AppImage from the terminal, I saw this message:
> It looks like you may be using a sandboxed version of FreeCAD. If OpenSCAD execution fails to load the temporary file, use FreeCAD's OpenSCAD Workbench Preferences to change the transfer mechanism.

So I went to
`OpenSCAD Workbench -> Preferences -> OpenSCAD -> Transfer directory`
There the current directly was I believe `/usr/bin` or something, so I changed it to some temporary directory I just created for testing this (because I was desperate to get it to work). And so that seems to have done it.

luzpaz, 2024-08-18:
@adrianinsaval perhaps there can be a more visible queue for flatpak users regarding this ? Some sort of dialog that can instruct them (maybe the way the CFDoF folks do it?)

ruffson, 2024-08-18:
@luzpaz Just for the record, I was never able to get the flatpak version to run. I have (and always had) the correct openscad version configured in the openscad preferences and the workaround with setting a transfer directory only worked on the weekly appimage version, but not on the flatpak. 

As originally stated, I gave the flatpak full system path accessibility, so this sandbox aspect of flatpak should not be at issue here.
For me it is unclear why (or what) is broken on the flatpak version? Maybe it is using the flapak openscad to open some inaccessible file but 

The weekly looks great by the way, font size is tiny but overall it gives off a very nice impression, great job!

luzpaz, 2024-08-18:
@adrianinsaval would you mind transferring this ticket to the FreeCAD-flatpak repo ?

luzpaz, 2024-08-18:
@ruffson you were never able to get the flatpak dev version to run ?

ruffson, 2024-08-30:
> @ruffson you were never able to get the flatpak dev version to run ?

Sorry for late reply. BUt yes I also experience this on the dev version of flatpak (which is still 0.21.0?).

From about:
```
[code]
OS: KDE Flatpak runtime (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33639 (Git)
Build type: Release
Branch: (HEAD detached at 1580fe7)
Hash: 1580fe7c9696654803a93da8cf372f75e75149fa
Python 3.10.11, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.7.1
Locale: English/United States (en_US)
[/code]
```

To prove that I (tried to) install(ed) the beta:

```
╰─❯ flatpak list | rg freecad
FreeCAD org.freecadweb.FreeCAD  0.21.0.33639    beta    flathub-beta    system
```

luzpaz, 2024-08-30:
Yea, you're absolutely right, i was under the wrong impression and now understand that flatpak is limited to stable version.

Roy-043, 2025-04-30:
I just found out that the programmatic boolean functions work without OpenSCAD. So maybe this dependency can be avoided altogether?
`Mesh.show(msh_box.unite(msh_cyl))`

luzpaz, 2025-09-17:
@adrianinsaval how about transferring to flatpak repo issue queue  ?

FEA-eng, 2026-02-13:
> [@maxwxyz](https://github.com/maxwxyz) can you move issue to flatpak repo ?

ping @adrianinsaval

FEA-eng, 2026-05-16:
@adrianinsaval Could you move it to the flatpak repo?

FEA-eng, 2026-08-16:
Should be moved to the flatpak repo.

maxwxyz, 2026-08-16:
@adrianinsaval FYI