FreeCAD/FreeCAD #15928: [CAM] Unclear UI in import pre-processor selection

alexfacciorusso, 2024-08-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The dialog for the selection of a pre-processor for importing gcode is not very user-friendly, containing "example" preprocessors in the release folder and no description on what each pre-processor does.

### Full version info

```shell
OS: Windows 11 build 22635
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
```

### Subproject(s) affected?

CAM

### Anything else?

Copied from #15909, separated the bug from this UI issue.

### Step 1 - open the file
[image]

### Step 2 - dialog displayed

You can see the dialog having code-friendly names instead of user-friendly ones. Also, no description on what each script does.

[image]

### Suggested approach

Add a user-friendly name and description to each pre-processor script and substitute those in the UI. Also useful to change that dropbox into a nicely-laid list, containing the user friendly data. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

alexfacciorusso, 2024-08-18:
I can pick this up, feel free to assign this to me! :)

knipknap, 2025-05-19:
@alexfacciorusso Are you still working on this?

luzpaz, 2025-11-11:
@alexfacciorusso ping