FreeCAD/FreeCAD #15936: Extrusion outside the boundaries of the part results in an invisible solid

luisDias89, 2024-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I create a sketch on top of a solid and extend it beyond the face, it results in an invisible solid. It should be possible to visualize the solid, especially since I might want to place it next to the other and only join them at the end.

https://github.com/user-attachments/assets/1a2b46e3-83f4-40c9-9e03-9b061ffb3667

In version 0.21, it gave an error and said it didn't create two solids. However, I also don't think that's the best approach.

In SolidWorks, the creation of two objects that are not connected automatically generates two separate bodies
[image]

Yes, if a connection is added between the two objects in SolidWorks, they automatically merge into a single body.
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2024-08-18:
This is expected behavior. Multi solid bodies is currently an experimental feature. You can enable this by selecting the body object and enabling this using the property panel.

luisDias89, 2024-08-18:
[image]

Ok, that's it then. But you can only visualize it if you delete all the bodies I've already made, activate it, and extrude them again. Additionally, if this is not enabled, it should behave like the old FreeCAD and not allow extruding multi-bodies, with a small message informing that if you want to enable it, go to...

obelisk79, 2024-08-18:
@kadet1090 ?

kadet1090, 2024-08-18:
@bgbsww ? This seems more like issue with missing single rule checks after toponaming.

FEA-eng, 2025-09-27:
Currently, if the Allow Compound is disabled, there's a preview and error messages to enable Allow Compound plus "Tip shape is empty" and a generic pop-up (#24188). And you can't accept it. If Allow Compound is enabled, it just pads the sketch as expected. I'm not sure if there's anything left to do here. Maybe only something about the aforementioned single solid rule check vs TNP but I'm not sure if it's reproducible and how.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

closed 2025-09-27 by luisDias89