FreeCAD/FreeCAD #15938: Sketcher - 3 Point Arc Tangent Constraint Loss

sparker0594, 2024-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[Dev 38467] Sketcher's 3-point arc tool will not retain the final constraint of the 3rd point when it is a tangent constraint.
If the 3rd point is Point on object, the 3rd constraint is retained.

Stable 0.21.2 Version behaviour is correct, the tangent constraint will be retained.

Demonstration 38467 Tangent:
[image]

Demonstration 38467 Point on Object:
[image]

Demonstration Stable 0.21.2 Tangent:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git)
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.65
  * CurvedShapes 1.0.9
  * Curves 0.6.42
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Sketcher

### Anything else?

FreeCAD file, incase it's helpful.

[Pull Start Adapter Plate.zip](https://github.com/user-attachments/files/16648036/Pull.Start.Adapter.Plate.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-18:
@PaddleStroke @FlachyJoe FYI

sparker0594, 2024-09-14:
Just in case it's helpful, behaviour Is still Present in RC1.

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: https://github.com/FreeCAD/FreeCAD/commit/60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods:

    3DfindIT 1.2.0
    Curves 0.6.47
    sheetmetal 0.4.25

PaddleStroke, 2024-10-24:
This bug is fixed by https://github.com/FreeCAD/FreeCAD/pull/17417

PaddleStroke, 2024-10-24:
erratum this bug is not fixed. It apparantly does not happen in all cases. I found a case where this bug still happen.

PaddleStroke, 2024-10-24:
Can reproduce. This happens if the rest of the sketch is fully constraint. There's an undue message : 

> Autoconstraints cause redundancy. Removing them

Minimal file for reproduction : 
[3p bug.zip](https://github.com/user-attachments/files/17506752/3p.bug.zip)

PaddleStroke, 2024-10-24:
I had a look at it and the auto-constraints generated by `DrawSketchHandlerArc `appears to be correct.

It appears that `DrawSketchDefaultHandler::removeRedundantAutoConstraints` is doing the removal.
Apparantly `sketchobject->diagnoseAdditionalConstraints(autoConstraints)` is somehow finding the sketch to have redundant. So it looks like its coming from the solver. Part of the code I don't know at all. @AjinkyaDahale could you please check this issue out?

AjinkyaDahale, 2025-03-14:
Still relevant?

luzpaz, 2025-07-21:
bumping for visibility

luzpaz, 2025-07-27:
@sparker0594 can you reproduce on 1.1-dev ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

PaddleStroke, 2025-09-25:
The reason why the tangent disappear is because it is redundant.

[image]

This issue is actually the same as 
https://github.com/FreeCAD/FreeCAD/issues/10248
https://github.com/FreeCAD/FreeCAD/issues/13645
etc.

PaddleStroke, 2025-09-25:
Actually this is wrong. Tangent is only 1 DoF. This should not be redundant.
So it is not the same as the other I guess

PaddleStroke, 2025-09-25:
[image]

A simpler case can also wrongly detect tangent as redundant.

PaddleStroke, 2025-09-25:
After a long back and forth with our friend gemini, I think we identified the root issue based on my previous simpler case : 

There're two rules for the solver to follow:
1. Rule 1 (Coincident Centers): Center2.x = Center1.x and Center2.y = Center1.y.
2. Rule 2 (Tangent Circles): The distance between the centers must be equal to the absolute difference of their radii. Mathematically: distance(Center1, Center2) = abs(Radius1 - Radius2).

Now, let's see what happens when the solver tries to understand these two rules together.:

- Since Center1 and Center2 are the same point, distance(Center1, Center2) is zero.
- The Tangent rule (Rule 2) therefore becomes: 0 = abs(Radius1 - Radius2).
- This simplifies to: Radius1 = Radius2.

So, for this specific case, adding a Tangent constraint to two concentric circles is geometrically and mathematically equivalent to adding an Equal Radii constraint.

**Why the "Redundant" Error?**
This is the key part. Before the solver adds a new constraint, it performs a redundancy check. It asks: "Based on the sketch's current state, does this new rule add any information I don't already have?"
This check is performed by looking at the gradient (the partial derivatives) of the constraint equation.

1- The "Good" Case (It Works):

- You add the Tangent constraint when Radius1 is 5 and Radius2 is 10.
- The solver checks the Tangent rule: 0 = abs(5 - 10). This is 0 = 5, which is false.
- The gradient of this equation is non-zero. The solver sees this as a valid, new piece of information.
- It accepts the constraint and solves the sketch by changing Radius2 to 5. The sketch is now fully constrained and correct.

2- The "Bad" Case (It Fails and reports "Redundant"):

- You add the Tangent constraint when Radius1 is 5 and Radius2 is also 5 (or 5.00000000001, which is numerically very close).
- The solver performs its redundancy check at this exact moment.
- It looks at the Tangent rule: 0 = abs(5 - 5). This is 0 = 0, which is true.
- Crucially, the gradient of the constraint equation at this point is the zero vector. You can see this in the code for GCS::ConstraintTangentCircumf and its derivatives. A constraint with a zero gradient provides no directional information for the solver to make a change.
- The solver concludes: "This new rule is already satisfied in the current state and provides no new information or direction for solving. It is therefore redundant."

The "sometimes" behavior is entirely down to floating-point precision. Whether abs(Radius1 - Radius2) is considered "zero" depends on whether it's smaller than an internal tolerance (Precision::Confusion() or similar). Dragging the geometry can easily make the values cross this tiny threshold, making the behavior appear inconsistent.

Conclusion/summary : mathematical singularity when the 2 circles are concentric and tangent.
Fix: Not sure yet, still digging.