FreeCAD/FreeCAD #15939: Sketcher - Fully constrained element not updated post partial constraint removal

sparker0594, 2024-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[Dev 38467] When a fully constrained element has a constraint removed which will give it a degree of freedom, the graphical status of the element is not updated (still displays are the fully constrained colours indications). The sketch will update upon other sketch elements being manipulated (moved) or an attempt to move (click and drag a static fully-constrained element, and the sketch will refresh).

Stable 0.21.2 Version behaviour is correct, the sketch will refresh immediately.

Demonstration of 38467:
[image]

Demonstration of Stable 0.21.2:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git)
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.65
  * CurvedShapes 1.0.9
  * Curves 0.6.42
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Sketcher

### Anything else?

FreeCAD File, if helpful

[Pull Start Adapter Plate.zip](https://github.com/user-attachments/files/16648116/Pull.Start.Adapter.Plate.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sparker0594, 2024-09-14:
Just in case it's helpful, behaviour Is still Present in RC1.

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * 3DfindIT 1.2.0
  * Curves 0.6.47
  * sheetmetal 0.4.25

FEA-eng, 2025-09-17:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```