FreeCAD/FreeCAD #15941: <Exception> Access violation

Luciniki, 2024-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

13:47:18  Inizializzazione ambiente A2plus v0.4.65.
13:48:28  Base exception thrown for 'C:\Users\Luciano\AppData\Roaming\FreeCAD\Mod\A2plus/GuiA2p/Resources/ui/a2p_prefs.ui'
13:48:28  <Exception> Access violation

this happens when i open preferences:
edit -> preferences -> A2plus

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git)
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.65
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-18:
Please open the issue in the A2plus repository

closed 2024-08-18 by maxwxyz

luzpaz, 2024-08-18:
The repo is at https://github.com/kbwbe/A2plus