FreeCAD/FreeCAD #15943: Attach sketch to angled face is not allowed

macdroid53, 2024-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In 0.22 the attaching to a face is not allowed if face is not parallel to a base plane.
[image]

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38475 (Git)
Build type: Release
Branch: main
Hash: 95c514fca3190c7b56c40f2fc4005a7337f64a56
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * MnesarcoUtils 0.2.5
  * lattice2 1.0.0
  * Curves 0.6.44
  * ArchTextures
  * Nodes 0.1.36
  * OpenTheme 2024.8.15
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * freecad.gears 1.2.0
  * sheetmetal 0.4.24
  * Silk 0.1.5
  * kicadStepUpMod 11.2.7
  * DynamicData 2.62.0
  * workfeature-macro
  * fasteners 0.5.27
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

macdroid53, 2024-08-18:
This is explained in the forum thread: "...The issue can be explained by looking at the Surface of the face. Although it looks, and in fact is, planar, it is technically a BSplineSurface. The attachment does not work for such a face...."

The error message should be improved to explain this at least. If it is possible to discern that the face is a not what it appears to be.

chennes, 2024-08-18:
That is a change, right? It was not a BSplineSurface before?

macdroid53, 2024-08-18:
Not sure. If, in PD, you create a Wedge primitive, it is a Face and the attachment works fine.

maxwxyz, 2024-08-18:
Using an angle in the pad seems to create non-planar surface.

macdroid53, 2024-08-18:
Apparently it is known that attach can't happen to a BSplineFace, even though it may be planar.
Is it possible to tell if a BSF is planar?

sedatkrn, 2024-08-18:
i didn't see any problem stabil version freecad.

kisolre, 2024-08-19:
There is new comment in the thread that from 0 to 12 deg a plane is generated and after that it switches to bsplinesurface so attachment is not possible.

sedatkrn, 2024-08-22:
if you use sketcher wb. that issue not see. you can tray.

sedatkrn, 2024-09-27:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38830 (Git)
Build type: Release
Branch: main
Hash: b38c42262f58cf0587b627e7a866d3756da34dc4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.67
  * Assembly3 0.12.2
  * BillOfMaterials 0.1.4.3
  * Curves 0.6.47
  * dodo 1.0.1
  * Estimate 0.1.2
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Plot 2022.4.17
  * POV-Ray-Rendering
  * Render 2024.8.4
  * sheetmetal 0.4.26
  * SteelColumn
  * WB_Organizer 2024.1.29
  * woodworking 0.21.2.33771

The issue is contuning 
[image]

maxwxyz, 2025-09-17:
Please test on the latest weekly

FEA-eng, 2025-09-17:
> Please test on the latest weekly

For me, it seems to be fixed there unless I'm missing some steps to reproduce it (but I recall recent fixes from werner aimed at this).

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

kisolre, 2025-09-17:
Not resolved. Here is an example file that fails under -20 and over 18 deg. With todays weekly.

[image]

[Example for fail.FCStd.zip](https://github.com/user-attachments/files/22391325/Example.for.fail.FCStd.zip)

```
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Bulgarian/Bulgaria (bg_BG)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.68
  * AnimationFreeCAD 1.0.0
  * ArchTextures
  * Assembly3 0.12.2
  * Assembly4 0.60.2
  * BIM021 2021.12.0
  * BOLTSFC
  * CadbaseLibrary 3.0.0
  * CadbaseLibrary-backup-before-zip-migration 2.0.0 (Disabled)
  * CubeMenu
  * CurvedShapes 1.0.13
  * CycloidGearBox
  * Design456 0.0.1
  * DesignSPHysics 0.8.1(29-05-2025)
  * dodo 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.41
  * fcgear 1.0.0
  * FEMbyGEN 2.5.1
  * FEM_FrontISTR 0.2.0
  * frame 0.1.1
  * freecad.gears 1.3.0
  * freecad.gears-backup-before-zip-migration 1.3.0 (Disabled)
  * freecad.xray 2022.4.17
  * FreeCAD_mcm_addon
  * Help 1.0.3
  * InvGears 0.1.2
  * Launcher
  * MakerWorkbench 1.0.1
  * Mechatronic
  * MeshRemodel 1.10.38
  * Movie 2025.1.4
  * Nikra-DAP
  * NikraDAP 2.0.0
  * Nodes 0.1.36
  * nurbs
  * OpticsWorkbench 1.2.2
  * ose-piping
  * PlacementTools
  * Plot 2024.11.26
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * Reinforcement
  * Render 2024.12.15
  * Rocket 4.1.1
  * SelectorToolbar
  * sheetmetal 0.7.53
  * Ship 2024.11.26
  * Silk 0.2.0
  * SlopedPlanesMacro
  * StemfieWB
  * symbols_library
  * WB_Organizer 2024.1.29
  * workfeature
  * workfeature-macro

```