FreeCAD/FreeCAD #15948: Spreasheet configuration table resets the value when selection string changes

nazar-pc, 2024-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I use string formatting from the actual value in configuration table, such that it is easier for the user to have a reference of what the end values should be.

The problem is that in FreeCAD selection seems to happen based on that string instead of a cell number (which I can see from Python console), which in turn means that in case values themselves are derived, they will be reset when something changes, which is not how I expected things to work, I expected for the selected cell to remain selected regardless of its changing text.

Here is a demo:

https://github.com/user-attachments/assets/2caf3856-4c24-4f80-90c4-a28f04112cb5

This is a really annoying issue that doesn't really have a decent workaround except avoiding formatting completely.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git) AppImage
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * parts_library
  * Curves 0.6.42
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-04:
@nazar-pc can you reproduce on weekly ?

nazar-pc, 2024-12-08:
Yes, still happening exactly the same on latest weekly:
```
OS: Ubuntu 24.04.1 LTS (Custom/custom/xcb)
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * parts_library
  * Curves 0.6.42
```

nazar-pc, 2024-12-08:
Unfortunately FreeCAD also didn't detect that something has changed either:

https://github.com/user-attachments/assets/2fd65638-0e36-4a90-9201-c756686cc183

nazar-pc, 2025-07-23:
Still an issue, tested in the latest weekly:
```
OS: Ubuntu 24.04.2 LTS (Custom/custom/xcb)
Architecture: x86_64
Version: 1.1.0dev.42653 (Git) AppImage
Build date: 2025/07/21 16:35:39
Build type: Release
Branch: main
Hash: c99cd73cd60874f231bda6233477e9051afcea57
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Ukrainian/Ukraine (uk_UA)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/83.3641/1.25
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * Curves 0.6.42
```

FEA-eng, 2026-01-30:
@nazar-pc Can you confirm if it's still relevant and share the file?