FreeCAD/FreeCAD #15951: Additive pipe not working.

GAZ082, 2024-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi. Attached macro, file and here is the video.

https://github.com/user-attachments/assets/fbbd5142-d040-4992-8d2d-a6ae6c9093e6

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git) AppImage
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Curves.backup1713713759.0032098 0.6.29 (Disabled)
  * QuickMeasure 2022.10.28
  * ExtremeProDark 2.9.0
  * Curves.backup1693059780.2922277 0.6.13 (Disabled)
  * CfdOF.backup1710694906.6805208 1.24.9 (Disabled)
  * Assembly4.backup1696800837.713216 0.50.2 (Disabled)
  * Curves.backup1714093860.7275996 0.6.34 (Disabled)
  * OpenTheme 2024.8.17
  * CfdOF.backup1699129791.898633 1.24.7 (Disabled)
  * Assembly4.backup1710902627.637761 0.50.9 (Disabled)
  * Defeaturing.backup1710696629.0288966 1.2.2 (Disabled)
```

### Subproject(s) affected?

PartDesign

### Anything else?

[Iman Puerta Terraza.FCStd.zip](https://github.com/user-attachments/files/16651195/Iman.Puerta.Terraza.FCStd.zip)
[macro.txt](https://github.com/user-attachments/files/16651210/macro.txt)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-08-21:
@luzpaz there's two scenarios in play, one is education in that there is a live vertex at the far end of the arc (away from the circle section sketch) for the Pipe path which shouldn't be there but there's no feedback to the user regarding this (maybe the code just ignores it).

But far more importantly the performance degradation for such as simple Additive Pipe is bordering on the unusable. I can perform way more complex Additive Pipes through multiple complex sections with near instantaneous response such as this Hook:

[image]

So the next step is to see whether I can at least narrow down to hopefully a few hundred commits when this performance degradation started and from there a Git Bisect to get a specific commit, whichever way it's going to be a while.

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38498 (Git)
Build type: Release
Branch: main
Hash: 7e311ed4b96299c8b9909d86cda2410a53cbd498
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * freecad.gears 1.2.0
  * Curves 0.6.44
```

GAZ082, 2024-08-21:
@Syres916 good call about having a live node, an error should pop up. However, i also had issues with another circle at the end, I'll see if can try later tonight.

Syres916, 2024-08-21:
@GAZ082 a little more education, you appear to have set in _Preferences > Part/Part Design > Shape view_ way too low Tessellation settings, here are mine which produced IMHO good enough screen representation but excellent performance:

[image]

I've adjusted the Body and two sketches in the attached file.

[Iman.Puerta.Terraza_Syres.FCStd.zip](https://github.com/user-attachments/files/16700695/Iman.Puerta.Terraza_Syres.FCStd.zip)

I definitely remember a code change in the past year that was supposed to alert users who changed their tessellation settings beyond a certain threshold and alerted them to this fact. I'll have a look in the morning to see if the thresholds are not high enough, it's difficult because if a user has a very recent top level processor with good single core throughput they would tolerate a much lower setting than my 12 year old i5 4 core CPU.

Syres916, 2024-08-21:
> I definitely remember a code change in the past year that was supposed to alert users who changed their tessellation settings beyond a certain threshold and alerted them to this fact. 

Found it, https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Part/Gui/DlgSettings3DViewPartImp.cpp#L65-L69
Maybe the angle setting needs the same treatment.

GAZ082, 2024-08-21:
@Syres916 I feel humbled by so much education **bows**

Yes, the tessellation was definitely the issue. Used your settings and rendered instantly! Do you still believe there is a underlying performance issue with the engine or this is just a rendering issue?

GAZ082, 2024-08-22:
@Syres916 i think the performance issue is still there. Now with even the lower tessellation I was forced to quit FC. Looks like how you perform the operation influences the outcome.

Syres916, 2024-08-22:
> i think the performance issue is still there. Now with even the lower tessellation I was forced to quit FC. Looks like how you perform the operation influences the outcome.

Firstly did you go back to your original file but without changing the _Max Deviation_ and _Max angle deflection_ on the Body and the two sketches, all three have an impact?

Changing the _Preferences_ only applies to new objects, existing files with different settings are not updated by design otherwise if a customer was collaborating with a supplier each time one or the other opened the shared file it would be constantly changing which is not helpful.

Before starting an _Additive Pipe_ I always select the _cross section sketch_ or _face_ first and then the _Path sketch_ before clicking the _Additive Pipe_ icon but it is designed to also have the objects selected after starting the operation.

GAZ082, 2024-08-22:
> Firstly did you go back to your original file but without changing the _Max Deviation_ and _Max angle deflection_ on the Body and the two sketches, all three have an impact?

I just changed the body, never realized the sketches also have this property. Will do more tests tonight taking in consideration your hints, thanks!

GAZ082, 2024-12-07:
This is working fine in 1.0. Closing.

closed 2024-12-07 by GAZ082