FreeCAD/FreeCAD #15952: No properties on Part WB chamfer

nazar-pc, 2024-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The only visible properties of Chamfer are Base, Placement and Label. Even when enabling hidden properties I do not see those that correspond to the length(s).

The reason I was looking there in the first place is because many tools in FreeCAD (for example array tools in Draft WB) do not have expression editor in their dialog, but properties can still be assigned to expressions in the properties panel. Something about chamfer implementation must be substantially different from many other tools.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git) AppImage
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * parts_library
  * Curves 0.6.42
```

### Subproject(s) affected?

Part

### Anything else?

Not sure if this is a bug report or a feature request. Also likely applies to fillet, though I didn't try it.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

nazar-pc, 2024-08-19:
I can edit with double-click, but as mentioned I can't use expressions there to reference spreadsheet values

Roy-043, 2024-08-19:
The information is stored in the "Edge"s property. It is hidden by default, but does appear in the Property editor after checking "Show hidden".

nazar-pc, 2024-08-19:
Hm... indeed, it is there. A few observations:
* first of all, it can't be edited, even though it it not read-only
* I do find it a bit confusing that one chamfer allows to create chamfers of different sizes on different edges, this makes little sense to me and seems inconsistent with other tools
* the format of `Edges` is different from other tools, I recall for some others it was just an array/list of values, but here it is a tuple of 3 elements, first of which seems to be the same as edge number in `Edge Links` property, which is awkward

So all in all, it doesn't help me with expressions much or anything at all really.

Roy-043, 2024-08-19:
You can just change the Edges property. The EdgeLinks property will update automatically if you do.

An expression with this basic syntax works:
`list(tuple(12; 1; 1), tuple(10; 1; 1))`

nazar-pc, 2024-08-19:
I already said above the property `Edges` is can't be edited, not with UI at least

Roy-043, 2024-08-19:
Not sure what you mean. You can assign an expression through the UI.

nazar-pc, 2024-08-19:
As I said, it is literally not editable at all:

https://github.com/user-attachments/assets/4c09e74a-2b10-4374-95cf-b0bedf3b1a51

nazar-pc, 2024-08-19:
Hm... that is the first time I see it like that. Usually there is an expression icon and I can press `=` to start editing. Everything seems to be different and awkward about this Chamfer :thinking: 

For example in array tools from Draft workbench I can assign expression to individual offsets instead of crafting the whole list manually, not to say it'll break immediately when I change underlying topology if I format that list with edge numbers manually.