FreeCAD/FreeCAD #15961: Unable to move section along X axis

kaiwas, 2024-08-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The section created on the basis of one of the views cannot be moved horizontally. No settings help.

At the same time, there are values ​​that allow you to move the section on the sheet. But the movement is reset if you move the section with the mouse.

Demonstration on video. 

https://github.com/user-attachments/assets/744acb10-4eab-4a64-92ce-e6f674d5cc7b

Remove the zip extension.
[Test_techDraw.FCStd.zip](https://github.com/user-attachments/files/16657276/Test_techDraw.FCStd.zip)

### Full version info

```shell
0,22dev
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-19:
@kaiwas can you please add the Full About info to the ticket ? TIA

kaiwas, 2024-08-19:
> can you please add the Full About info to the ticket ? TIA

I added information about the current state of the system (is that enough?)
But I noticed this problem more than a month ago.

What is TIA?

luzpaz, 2024-08-19:
> I added information about the current state of the system (is that enough?)
But I noticed this problem more than a month ago.  

@kaiwas it's just good housekeeping to do so. Sometimes tickets get missed and in this case it would save triage time by knowing what specific revision you tested it on.

> What is TIA?

Thank in advance :grin:

WandererFan, 2024-08-19:
The fix for this is included in queued PR #15450.

kaiwas, 2024-08-19:
> The fix for this is included in queued PR #15450.

Good news. Thank you!

luzpaz, 2024-08-28:
Weird, I can't link this ticket to the PR

PR #15450 (2024-09-02): [TD]Fix section snapping (fix #15961)
This PR implements a fix for the section view snapping algo.  It also fixes #15312.
commit e852052d (2024-09-02): [TD]Fix section snapping (fix #15961) (#15450)

* [TD]Add view snapping preferences

* [TD]fix section snapping algo

- snap sections to section normal line.
- snap views to other views in X&Y

* [TD]fix snapping to ProjectionGroups
closed 2024-09-02 by yorikvanhavre