FreeCAD/FreeCAD #15962: [TexhDraw] There is no way to move one of the views (consisting of a projection group) along the "secondary" axis

kaiwas, 2024-08-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sometimes you may need to place views on a sheet in any order. But there is no way to "turn off" the snapping of a view relative to the "secondary" axis.

[image]

Remove the zip extension.
[Test_techDraw.FCStd.zip](https://github.com/user-attachments/files/16657570/Test_techDraw.FCStd.zip)

### Full version info

```shell
0.22dev
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-19:
IMO thats the intended feature of a projection group. If you need a separate view, just insert another single view.

kaiwas, 2024-08-19:
> IMO thats the intended feature of a projection group. If you need a separate view, just insert another single view.

I can only partially agree with you.
It is much more convenient to create many views in one place and "unbind" them than to create a view separately many times.

In the end, we are talking about one single setting that could disable the binding.

Or there could be some button that "destroys/divides" the group.

By the way, such a button is missing in many places. But instead, you have to run the same operation many times. It seems that users should like to perform the same repetitive tasks. Instead of doing one "batch" operation in one place and then splitting it.

sparker0594, 2024-08-19:
Surely if space is an issue on your drawing you should use a second sheet or change the scaling?
Breaking the link of a projected view, although possibly convenient (and I know it's a possibility on other CAD programs), defeats the purpose of a 'Projected' view.

Breaking projected views makes the drawing harder to read/interpret, especially if your drawing contains a reference to first or third-angle projection. Most engineers wouldn't expect a floating view, unless is it contained in a detailed view with references.

Is an independent single view that laborious? 
Maybe your use case is more demanding than I imagine.

kaiwas, 2024-08-19:
> Surely if space is an issue on your drawing you should use a second sheet or change the scaling? Breaking the link of a projected view, although possibly convenient (and I know it's a possibility on [...]

You are proceeding from the approach that you are "modeling" a part (or something like that).
In architectural and construction drawings, the arrangement of "subviews" does not have a mandatory orientation rule.

In addition, I pointed out that such a setting should not be the default. It would be better if it could be activated when necessary. It would still be much faster than creating each view separately.

Well, do we really need to conduct tests and demonstrations to show that this is really convenient?

kaiwas, 2024-08-19:
This discussion reminds me of the discussions that were held on "separate/unrelated" parts in part design. Many said that such a possibility is harmful, because it does not correspond to physical processes and the user can make a mistake.
But this limitation is essentially artificial. In the end, the possibility of separate parts was introduced, which is actually correct. Because it will speed up some of the tasks.

Let me give you another example. I can quickly arrange columns in an array. But to make the columns unique - I will have to copy a separate column. Paste it and move it in space. Instead, I could "break" the array (but there is no such possibility). And there are many, many such examples. I have to use workarounds everywhere.

kaiwas, 2024-08-19:
Quick test of creating several views.
Each time I had to select the scale.
As a result, the time difference is about 3 times.

(it was the creation that was tested. without placing it on the sheet. In addition, before this I made several attempts trying to make the creation of separate views as quickly as possible. Without this training, I spent even more time)

https://github.com/user-attachments/assets/d0cdca36-ab96-499d-8693-ec4e9c5da860

kaiwas, 2024-08-19:
By the way, it would also be useful to set an arbitrary scale for any of the subsections. But in the current logic of the group of views, this is logically impossible (because they all have to be the same size and arranged in a certain order).

The whole problem is that the process of creating individual views is very complicated. If there are a lot of objects, they all have to be selected before the section operation, and so on for each of them.
If everything was done at once and then "adjusted" it would greatly simplify some tasks.

sparker0594, 2024-08-19:
Understood, I will admit, that I only had Parts designing in my head, I did not realise the frequency of this application in the Architecture designs.

WandererFan, 2024-08-19:
You can set a default scale to be used for all newly created views in preferences, no need to set the same scale individually.

kaiwas, 2024-08-19:
> You can set a default scale to be used for all newly created views in preferences, no need to set the same scale individually.

Yes. Thank you. I know that.
There really is such a possibility.
It's a pity that in fact you often have to work with different scales (depending on the size of the object and the size of the sheet).

Let's say in 2 projects I use the main scale of 1:100, and in the third 1:50.
It would be convenient to set the scale once for the group and then change the scale of one of the elements if necessary. And not open the global settings.

In general, this is another workaround (limitation).

maxwxyz, 2024-08-19:
Any input to the original issue or can I close this?

kaiwas, 2024-08-19:
> Any input to the original issue or can I close this?

Why should you close it? Just because there are workarounds?
My opinion will remain the same - it can be done better. But you can leave it as is and use workarounds (as usual).

kaiwas, 2024-09-27:
> Any input to the original issue or can I close this?

You can close it. It seems humanity is not ready for such changes yet )))
We will return to this issue sometime later. In the meantime, there are more serious problems.

closed 2024-09-27 by maxwxyz