FreeCAD/FreeCAD #15963: [TechDraw] The identifier of the main view when creating a projection group.

kaiwas, 2024-08-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I suggest introducing the main view identifier (for example, a navicube analogue).
The current version with vectors is very difficult to understand the direction. But it can certainly be left and just add a visual auxiliary element.

A possible concept is in the picture.

[image]

You can develop this theme and add arrows with turns in addition to the cube itself (as is currently done for the 3D view). But there are concerns that these arrows will not fit on the panel.

### Full version info

```shell
0,22dev
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-19:
@kaiwas Would you mind making screenshot of a how it currently looks so the ticket presents a 'before and after' template.  I like this idea btw

kaiwas, 2024-08-19:
> Would you mind making screenshot of a how it currently looks so the ticket presents a 'before and after' template.

I updated the first message. Thanks!

obelisk79, 2024-08-21:
Seems like a reasonable proposal. I'll need time to kill this over, in either case this would be a post-1.0 release item.

WandererFan, 2024-10-30:
What shou