FreeCAD/FreeCAD #15964: I got startup error

budiz1, 2024-08-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When i start the latest Freecad weekly build, i get an error that says (in the picture below).

I dont know if that is my computers problem or freecads as this error never showed up before.

[image]

Thank you for your feedback,
Budiz

### Full version info

```shell
I cant enter Freecad
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-19:
@JohnOCFII any idea how to solve this ?

JohnOCFII, 2024-08-19:
> @JohnOCFII any idea how to solve this ?

Sadly, no.  I just downloaded the latest build (38467) for macOS arm -- and it comes across as damaged.  I know that there were efforts over the weekend to change the way the bundle was created.  Wonder if it is related to that change?  See https://github.com/FreeCAD/FreeCAD-Bundle/issues/290
[image]

oursland, 2024-08-19:
@JohnOCFII Yes, this is likely due to #290.

Syres916, 2024-08-19:
@oursland I think you meant to link https://github.com/FreeCAD/FreeCAD-Bundle/pull/290 rather than a FreeCAD PR from September 2016!!

oursland, 2024-08-19:
@adrianinsaval When I try to debug the application it dies due to signal 9 (SIGKILL), which indicates an external process is terminating the execution.

[image]

Is this perhaps a signing issue?

oursland, 2024-08-19:
@Syres916 Yes, GitHub automatically linked my text and did so incorrectly.

oursland, 2024-08-19:
I do believe it has to do with code signing or is somehow related.  I cannot manually launch the `freecad` binary.  When I move the file outside the `FreeCAD.app` directory, they begin to execute normally.

I did test this change extensively on my laptop for both https://github.com/FreeCAD/FreeCAD-Bundle/pull/290 and https://github.com/FreeCAD/FreeCAD-Bundle/pull/289, but I will admit I am less knowledgeable about macOS's requirements for running code outside the developer's machine and the related code-signing issues.