FreeCAD/FreeCAD #15966: FreeCAD.closeDocument(name) if name contains minus symbol ie "-"

spanner888, 2024-08-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Code below reproduces. Found trying write macro requested by @sliptonic to update all CAM Tool shape files, where error occurred with both "v-bit" and "thread-mill" shapes.

Also possible related is [[V-bit problem with 3Dsurface](https://forum.freecad.org/viewtopic.php?t=76665#p666147)](https://forum.freecad.org/viewtopic.php?t=76665)

`def file_test(name):
	FreeCAD.newDocument(name)
	FreeCAD.ActiveDocument.saveAs("/home/spanner888/Documents/" + name)
	FreeCAD.closeDocument(name)
	
f_names = ["something", "some_thing", "some-thing"]

for f in f_names:
	file_test(f)
	print("finished: ", f)`

`21:08:14  finished:  something
21:08:14  finished:  some_thing
21:08:14  pyException: Traceback (most recent call last):
  File "/home/spanner888/.local/share/FreeCAD/CamScripts/minus_in_filename.FCMacro", line 10, in <module>
    file_test(f)
  File "/home/spanner888/.local/share/FreeCAD/CamScripts/minus_in_filename.FCMacro", line 5, in file_test
    FreeCAD.closeDocument(name)
<class 'NameError'>: Unknown document 'some-thing'`

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

spanner888, 2024-08-19:
Was going to code a workaround and found below for a test document  "Un-named" and saw below in FC python console:

`>>> # Gui.Selection.addSelection('Un_named','Sketch')
>>> App.getDocument('Un_named').recompute()
>>> ### Begin command Std_Save
>>> # Gui.SendMsgToActiveView("Save")
>>> App.getDocument("Un_named").save()
>>> ### End command Std_Save
`
Also tried:

`>>> App.ActiveDocument.Name
'Un_named'`

Image shows the file is saved/named with the minus symbol:
[image]

This seems to show the auto change from minus to underscore is by design. Looked at FC wiki File Save/As etc no info.

I searched broader for "valid FreeCAD filename" etc etc, but not likely info, some references to issues with other character sets, but felt those not relevant.

spanner888, 2024-08-20:
And from FC python console:
`>>> FreeCAD.openDocument('/home/spanner888/Documents/_source/_APPS/FC_wkly-38459/squashfs-root/usr/Mod/CAM/Tools/Shape/v-bit.fcstd')
<Document object at 0x55ff20339160>
>>> # App.setActiveDocument("v_bit")
>>> # App.ActiveDocument=App.getDocument("v_bit")
>>> # Gui.ActiveDocument=Gui.getDocument("v_bit")`

So change appears to be from FreeCAD.openDocument(), not the closeDocument().

This could be as expected, except that the doc tree still has the "-", not "_" name and user is not informed of automatic/enforced change.

Other side effects, if you save document from toolbar it has the underscore name, then later user might open older doc with minus name.

luzpaz, 2024-08-20:
@spanner888  JFYI, if you want to display codeblocks you need to use 3 backticks on a separate line at the beginning and at the end of the block. I edited you the ticket to fix the formatting issues.

jkvfc, 2024-08-21:
I was also challenged by @sliptonic 's hint in #15902 therefore the attached macro.
[updateFilesTNP.FCMacro.zip](https://github.com/user-attachments/files/16690856/updateFilesTNP.FCMacro.zip)
[updateFilesTNP.zip](https://github.com/user-attachments/files/16690859/updateFilesTNP.zip)
Files and directories can be chosen. What do you think about it? Thank you for a quick look at it. I'm a complete rookie.
What should be done if this solution might be useful for others?

@spanner888 If in my solution lines 37,39,42 are deactivated and lines 38,43 are activated, I have the same issue as described here.

jkvfc, 2024-08-21:
@sliptonic I forgot to generalise some paths therefore a new version of the above mentioned script 
[updateFilesTNP.FCMacro.zip](https://github.com/user-attachments/files/16700476/updateFilesTNP.FCMacro.zip)

spanner888, 2024-08-21:
@jkvfc Sorry only looked at first version of your macro, so at least part of my generalised directory comment below no longer relevant.

Your macro does get past the save issues I had with v-bit & thread-mill, with a few minor issue (see further below).

It also caused me to realise there were more scenarios to cater for:
- Older "Mod/Path/Tools/Shape/" vs "Mod/CAM/Tools/Shape/" - this just needs code to detect which directory.
- AppImage and possibly other portable versions are not directly writable. 
    The AppImage can be decompressed using filename --appimage-extract, then cd squashfs-root/ & run AppRun, then the macro. However I have no idea about the other windows portable or Mac......
    So all I can sugegst is a note to users running AppImage and also hope someone else knows about the other possibilites.
- There is a general expectation that users should keep their version of FreeCAD up to date, which will eventually replace the files with already recomputed/saved versions. The forum shows many users continue to run older versions for many reasons, but this issue does not occur in current stable, eg: Version: 0.21.2.33771 (Git) AppImage and only in the weekly builds since then.

So with the fix already committed, I no longer think this is worth any effort.

    
Suggested changes to your script:

1. Use built in features to point both file and directory buttons to where the system shape files should be, which also removed your Linux (Mac?) only directory paths that would only work on your computer.

2. Also those system shape files are all lower case ".fcstd" so another change for the file open, otherwise in Linux cannot select any files.

3. There was also a bug in (*.FCStd") as it was missing another bracket inside the quote ie (*.fcstd)").

4. Need document discussion above or auto detect if filepath is Mod/Path or Mod/CAM.