FreeCAD/FreeCAD #15971: pad makes a copy sketch's attachment

aavogt, 2024-08-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the attached motor-beam-bug.FCStd,

1. make the last body active
2. select Sketch17
3. pad

This is what I get:

[pad-becomes-attached-translated.webm](https://github.com/user-attachments/assets/208560df-111d-43b7-ac04-af5ab7f571f5)

I expected to get a rectangular prism above or below the sketch. Instead the "Pad" is a translated copy of the BRACKET FRONT that Sketch017 is attached to.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (xmonad)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git) AppImage
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Canada (en_CA)
Installed mods: 
  * lattice2 1.0.0
  * freecad.gears 1.0.0
  * FEMbyGEN 1.2.0 (Disabled)
```

### Subproject(s) affected?

PartDesign

### Anything else?

[motor-beam-bug.zip](https://github.com/user-attachments/files/16663902/motor-beam-bug.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Flavelius, 2025-08-22:
I had the same happen to me:

https://github.com/user-attachments/assets/21338a8b-952f-4f05-84d1-9084278d08d7

```
OS: KDE Flatpak runtime (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 256fc7e)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * Curves 0.6.68
  * Dracula 0.0.9
  * Manipulator 1.6.0
  * freecad.gears 1.3.0
  * OpenTheme 2025.5.20

```

FEA-eng, 2026-09-20:
Still relevant (only the preview is correct)

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.8854/Intel/Intel(R) UHD Graphics 620
```

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/f151cbaf30ce8eac803811b03661f1c943dbcd62

The first Pad no longer pulls in the shape that its sketch is attached to in another Body. Could be picked up for a PR.