FreeCAD/FreeCAD #15972: "Save As" duplicates "FCStd"

ChuckInGame, 2024-08-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

it's not a major one, sure.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git) AppImage
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Render 2024.8.4
  * ThreadProfile 1.90.0
  * Help 1.0.3
  * Assembly4.backup1723411334.8566418 0.50.15 (Disabled)
  * freecad.gears 1.2.0
  * AirPlaneDesign 0.4.0
  * 3D_Printing_Tools
  * Assembly4 0.50.15
  * fasteners 0.5.26
  * MeshRemodel 1.9.23
  * pcb 6.2023.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-08-19:
Duplicate of #15908 and already fixed.

closed 2024-08-19 by Roy-043

ChuckInGame, 2024-08-19:
thanks, sorry that i did not found it