FreeCAD/FreeCAD #15974: Undocking Clipping-Bar: Does not reopen on the old position after FreeCad restart like Tasks-Bar or Models-Bar

ChuckInGame, 2024-08-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I can detach the tasks-bar and the model-bar from the FreeCad window, and after restart, FreeCad opens these Toolbars on the same position where they was when closing.

That's very useful and works fine.

I want to use the clipping-bar in the same way, but it does not work :(

-> each restart, the clipping-bar is closed again an I need to open it
-> when I open the clipping-bar, its docked in the FreeCad window.

Would be nice, if every toolbars can be used like the tasks-bar or the model-bar on multiple monitors.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git) AppImage
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Render 2024.8.4
  * ThreadProfile 1.90.0
  * Help 1.0.3
  * Assembly4.backup1723411334.8566418 0.50.15 (Disabled)
  * freecad.gears 1.2.0
  * AirPlaneDesign 0.4.0
  * 3D_Printing_Tools
  * Assembly4 0.50.15
  * fasteners 0.5.26
  * MeshRemodel 1.9.23
  * pcb 6.2023.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-24:
What is the clipping bar?

luzpaz, 2024-08-24:
@ChuckInGame can you provide a screenshot or even better a screencast ?

ChuckInGame, 2024-08-25:
sorry for the delay,

here is a recording:

https://github.com/user-attachments/assets/099b6f5e-4bfd-4ef8-aa60-c907374512d6

ChuckInGame, 2024-08-25:
by the way, could you please reduce the spaces in the axis sections. Currently the window is very big for the simple settings

ChuckInGame, 2024-12-03:
issue tested with :

OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39315 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 710542d121c2cabce05561591cfb00ef3b6e745c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default

Problem still exists, unchanged.

ChuckInGame, 2024-12-20:
Still reproduceable:

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39646 (Git) Conda
Build type: Release
Branch: main
Hash: b25b83c1a58bf0ae95cbd502d0ccae27e8127cad
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.33
  * freecad.gears 1.3.0