FreeCAD/FreeCAD #15978: Confusing and hard to find Tree View settings.

byteofsoren, 2024-08-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When searching for settings for the tree view you will find that those settings are scattered around in the settings tree under random topics. Example

- Display->UI has tree view settings.
- Display->Advanced has tree view settings like font size and padding.

And I think I have found tree view settings on other places to but could right now find them.
So how about creating a tree view branch under the Display tree and collect all those settings in that tree view branch?

```
Display
  |- Tree View
```
I think that would make things easier.

Also mention that Settings->Display->UI->Display settings do not work as mentioned in #15873

### Full version info

```shell
OS: Arch Linux (awesome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git) AppImage
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jangalasriramd7, 2024-09-22:
Hi there, can i work on this issue

byteofsoren, 2024-10-07:
@jangalasriramd7 Sorry for being absent. Lots of other things to do.
For me, you can absolutely work on this issue.