FreeCAD/FreeCAD #15979: [TechDraw] Regression from PR15935, some curved geometry becomes distorted

Syres916, 2024-08-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Unfortunately in fixing Issue #15901 the fix https://github.com/FreeCAD/FreeCAD/commit/558db30497800c765a81d9bb3f4b34158e43fa23 has caused a regression. Instead of the Techdraw geometry of the attached file looking like:

[image]

it appears:

[image]

I've made a feeble attempt to understand why some B-splines use float absolute to give the correct geometry and others don't but I'm non the wiser sorry to say.

Forum discussion: https://forum.freecad.org/viewtopic.php?t=89914

Just rename the extension from _zip_ to _FCStd_:
[SquareToRoundPipe.zip](https://github.com/user-attachments/files/16679857/SquareToRoundPipe.zip)

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git)
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * freecad.gears 1.2.0
  * Curves 0.6.44
```

### Subproject(s) affected?

TechDraw

### Anything else?

While this may not be a 1.0 Blocker, I certainly think it's quite high priority to cover off as many of the test files from this issue and the previous one if at all possible.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-08-21:
@WandererFan purely brainstorming here, but should the camera view direction be compared to the normal of the edge and if they match then use the code without _fabs_ otherwise use _fabs_?

Syres916, 2024-08-21:
@WandererFan it's crude and I'm sure there's a way to code it much more succinctly but this appears to work https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:TD_Handle_BSplines_Improvement

Syres916, 2024-08-21:
@luzpaz could you please add label for **Status: Solution Proposed** , TIA

Syres916, 2024-08-22:
> in SquareToRoundPipe and the file from #15901 the problematic edges are 2d projected edges and should both be on XY plane. This makes me think they should both take the same branch in your solution?

Maybe I'm not thinking correctly here but I thought the code I added:

```
    GeomLProp_CLProps propSpline(spline, f, 3, Precision::Confusion());
    gp_Dir gdir;
    propSpline.Normal(gdir);
    bool isSplineOnStdPlane = DrawUtil::isStdPlane(Base::Vector3d(gdir.X(), gdir.Y(), gdir.Z()));
```

is for returning 3D global space normal vectors, so when I add some basic debugging output of `gdir.X(), gdir.Y(), gdir.Z()`, for the __4D_49226__Hembrug_Chuck_Centre__with_drawing_ View I get:

```
15:57:15  X: -0.000000 Y: -1.000000 Z: 0.000000
15:57:15  X: 0.000000 Y: 1.000000 Z: 0.000000
```

hence does not use fabs but for just the _SquareToRoundPipe_ specifically _FrontTopRight_ view I get:

```
16:00:25  X: -1.000000 Y: -0.000000 Z: 0.000000
16:00:25  X: -0.511126 Y: 0.859506 Z: 0.000000
16:00:25  X: -0.866025 Y: -0.500000 Z: 0.000000
16:00:25  X: -0.500000 Y: 0.866025 Z: 0.000000
16:00:25  X: -0.731307 Y: -0.682048 Z: 0.000000
16:00:25  X: 0.082199 Y: 0.996616 Z: 0.000000
16:00:25  X: -0.082199 Y: -0.996616 Z: 0.000000
16:00:25  X: 0.731307 Y: 0.682048 Z: 0.000000
16:00:25  X: 0.500000 Y: -0.866025 Z: 0.000000
16:00:25  X: 0.488504 Y: -0.872561 Z: 0.000000
16:00:25  X: 0.866025 Y: 0.500000 Z: 0.000000
16:00:25  X: -1.000000 Y: 0.000000 Z: 0.000000
16:00:25  X: -0.509891 Y: 0.860239 Z: 0.000000
16:00:25  X: -0.327326 Y: 0.944911 Z: 0.000000
16:00:25  X: -0.000000 Y: 1.000000 Z: 0.000000
16:00:25  X: 0.755929 Y: 0.654654 Z: 0.000000
16:00:25  X: 0.832240 Y: 0.554416 Z: 0.000000
16:00:25  X: 0.489834 Y: -0.871816 Z: 0.000000
16:00:25  X: 0.327327 Y: -0.944911 Z: 0.000000
16:00:25  X: 0.000000 Y: -1.000000 Z: 0.000000
16:00:25  X: -0.755929 Y: -0.654654 Z: 0.000000
16:00:25  X: -0.832240 Y: -0.554416 Z: 0.000000
16:00:25  X: 1.000000 Y: 0.000000 Z: 0.000000
16:00:25  X: -0.500000 Y: -0.866025 Z: 0.000000
16:00:25  X: 0.512688 Y: -0.858575 Z: 0.000000
16:00:25  X: 0.500000 Y: 0.866025 Z: 0.000000
16:00:25  X: -0.485101 Y: 0.874458 Z: 0.000000
16:00:25  X: -1.000000 Y: 0.000001 Z: 0.000000
16:00:25  X: 1.000000 Y: 0.000002 Z: 0.000000
16:00:25  X: 0.500000 Y: 0.866025 Z: 0.000000
16:00:25  X: -0.486071 Y: 0.873919 Z: 0.000000
16:00:25  X: -1.000000 Y: 0.000002 Z: 0.000000
16:00:25  X: -0.500000 Y: -0.866025 Z: 0.000000
16:00:25  X: 0.512153 Y: -0.858894 Z: 0.000000
16:00:25  X: 1.000000 Y: 0.000019 Z: 0.000000
16:00:25  X: 0.996721 Y: 0.080913 Z: 0.000000
16:00:25  X: 0.997330 Y: -0.073031 Z: 0.000000
16:00:25  X: 0.511272 Y: -0.859419 Z: 0.000000
16:00:25  X: 0.995991 Y: -0.089458 Z: 0.000000
16:00:25  X: 0.996092 Y: -0.088318 Z: 0.000000
16:00:25  X: 0.996089 Y: -0.088351 Z: 0.000000
16:00:25  X: -0.488502 Y: 0.872563 Z: 0.000000
16:00:25  X: -0.486352 Y: 0.873763 Z: 0.000000
16:00:25  Geom::baseFactory - OCC error - GC_MakeCircle::Value() - no result - while making spline
16:00:25  X: 0.978184 Y: 0.207739 Z: 0.000000
16:00:25  X: -0.513446 Y: 0.858122 Z: 0.000000
16:00:25  X: -0.512781 Y: 0.858519 Z: 0.000000
16:00:25  X: 1.000000 Y: 0.000000 Z: 0.000000
16:00:25  X: 0.994505 Y: 0.104688 Z: 0.000000
[... 143 lines cut ...]
16:00:25  X: 0.067154 Y: 0.997743 Z: 0.000000
16:00:25  X: 0.074283 Y: -0.997237 Z: 0.000000
16:00:25  X: 0.067153 Y: -0.997743 Z: 0.000000
16:00:25  X: 0.064544 Y: -0.997915 Z: 0.000000
16:00:25  X: 0.074571 Y: -0.997216 Z: 0.000000
16:00:25  X: 0.984830 Y: -0.173523 Z: 0.000000
16:00:25  X: 0.990119 Y: 0.140233 Z: 0.000000
16:00:25  X: 0.074785 Y: 0.997200 Z: 0.000000
16:00:26  X: 0.984830 Y: 0.173521 Z: 0.000000
16:00:26  X: 0.984830 Y: -0.173521 Z: 0.000000
16:00:26  X: 0.984830 Y: -0.173523 Z: 0.000000
16:00:26  X: 0.984830 Y: 0.173521 Z: 0.000000
16:00:26  X: 0.067154 Y: 0.997743 Z: 0.000000
16:00:26  X: 0.074785 Y: 0.997200 Z: 0.000000
16:00:26  X: 0.990119 Y: 0.140233 Z: 0.000000
16:00:26  X: 0.067153 Y: -0.997743 Z: 0.000000
16:00:26  X: 0.984830 Y: -0.173521 Z: 0.000000
16:00:26  X: 0.074571 Y: -0.997216 Z: 0.000000
16:00:26  X: 0.064544 Y: -0.997915 Z: 0.000000
16:00:26  X: 0.067154 Y: 0.997743 Z: 0.000000
16:00:26  X: 0.067153 Y: -0.997743 Z: 0.000000
```
and so it does use fabs.

So when you say the XY Plane for the problematic edges in the two files I'm struggling to see why??

Syres916, 2024-08-24:
I'll study and hopefully learn from your PR, I have some models including a cylinder head with angle vale seats that maybe good tests, thanks @WandererFan.

WandererFan, 2024-08-24:
@Syres916 more tests would be great.

commit 884cf22c (2024-08-26): [TD]fix false positives in isCircle (fix #15979)
closed 2024-08-26 by yorikvanhavre