FreeCAD/FreeCAD #15983: [TexhDraw] When creating views and sections, doors and windows in the main object are not taken into account (automatically).

kaiwas, 2024-08-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The current behavior when creating architectural views is very inconvenient. I consider this point critical.
Now you need to select windows and doors so that they are displayed in views and sections. Selection at the time of creation is mandatory.
Also, if at some stage we add a new window or door - you need to add them manually to the view through an inconvenient menu.

Video demonstration.
https://github.com/user-attachments/assets/16da639a-46f8-4505-9d59-9a85e24c83a7

File for tests. Remove the zip extension.
[TD_View.FCStd.zip](https://github.com/user-attachments/files/16687104/TD_View.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git) AppImage
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * OpenTheme 2024.8.17
  * PitchedRoof
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-10-30:
related/duplicate: #15984

WandererFan, 2024-10-30:
mostly related to #15984:
How is the program to know that both the mirror and original objects should be used as the source?  If the original is automatically included, how does a user specify that only the mirror is wanted?

If you want to access both the original and the mirror, then create a compound of both first.

Not saying that this is necessarily the best ux, but it is consistent with Part WB.

WandererFan, 2024-10-30:
re doors and windows:
TD asks the Source object(s) for a shape.  If that shape (the building?) does not include some pieces (doors and windows), then they won't be drawn.  AFAICT, the shape of the building includes the cut-outs for doors and windows, but not the doors/windows themselves.

If there is a nice way to identify that a Source is an Arch object whose children(?) should be included, then we could probably modify the ShapeExtractor to special case this.

closed 2026-10-31 by maxwxyz