FreeCAD/FreeCAD #15984: TechDraw: Views and sections do not automatically include mirror objects.

kaiwas, 2024-08-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a view or section, child objects are not taken into account. Each time we perform this action, we need to manually add the desired object.

Also, views and sections do not take into account whether the object is hidden or not.

Video demonstration

https://github.com/user-attachments/assets/2d2f3408-65a2-4b45-add3-3a8ee555bffb

File for tests. Remove the zip extension.
[TD_mirror.FCStd.zip](https://github.com/user-attachments/files/16687734/TD_mirror.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git) AppImage
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * OpenTheme 2024.8.17
  * PitchedRoof
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-21:
Related to: #15983

kaiwas, 2024-08-21:
> Related to: #15983

Yes. The problems are similar. But I don't know if there is a difference in approaches between workbenches when constructing views. Perhaps such specificity manifests itself somewhere else where there are "child" objects. If you consider one of the messages unnecessary - then it is better to leave the version with windows and doors (the topic is linked), and this one can be closed and deleted.