FreeCAD/FreeCAD #15991: Regression: Undocked panels don't yield when other apps are selected in latest build

macdroid53, 2024-08-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In today's build, when first launched with un-docked panels (Property and Task) on another monitor, the panels yield focus as expected. After some operations (like running a Python addon) or errors are reported, the panels no longer yield when another  application is given focus on the monitor that has the panels.
This regression seems to happen every few months, checking how it was repaired previously may help determine the cause.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38498 (Git)
Build type: Release
Branch: main
Hash: 7e311ed4b96299c8b9909d86cda2410a53cbd498
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * MnesarcoUtils 0.2.5
  * lattice2 1.0.0
  * Curves 0.6.44
  * ArchTextures
  * Nodes 0.1.36
  * OpenTheme 2024.8.17
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * freecad.gears 1.2.0
  * sheetmetal 0.4.24
  * Silk 0.1.5
  * kicadStepUpMod 11.2.7
  * DynamicData 2.62.0
  * workfeature-macro
  * fasteners 0.5.27
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-21:
'Yield' as in they always stay on top ?

maxwxyz, 2024-08-24:
@luzpaz is this really a blocker for the release candidate? Can anyone confirm this issue? I cannot reproduce on Windows

luzpaz, 2024-08-24:
> After some operations (like running a Python addon) or errors are reported, the panels no longer yield when another application is given focus on the monitor that has the panels.

@maxwxyz this seems like a very disruptive UX issue in FC and for the persons' computer. I can't test ATM. If we can't reproduce then remove blocker status ? Or remove now and add when it's reproducible ? I defer to you.

macdroid53, 2024-08-24:
I'm not sure this email will work. I'm on my phone. I don't think this was
ever reproducible in windows. I ca reproduce in Debian and Manjaro every
time it regresses. If you can assume most users don't care about
unlocked/multi-monitor behavior, then it is obviously not a blocking issue.

bgbsww, 2024-08-25:
Huh.  Can't reproduce on a 3840x2160 primary display and a 3840x2400 secondary monitor connected via USB3/Firewire.

Tried BIM, Curves, OpenSCAD workbenches, lots of model modifications.  I'm in the "not a blocker until there's a reproduction path" camp, was hoping to reproduce, but no luck so far.

OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38501 (Git)
Build type: Debug
Branch: bgbsww-toponamingRevolution
Hash: 4ea2a0162d2acf74679864d2ce865e7fdd672f1b
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.4.24
  * Curves 0.6.45
  * DynamicData 2.62.0
  * BIM021 2021.12.0
  * parts_library

luzpaz, 2025-07-27:
@macdroid53 can you still reproduce ?

FEA-eng, 2025-09-27:
@macdroid53 Is it still ok? If yes, the issue should be closed.

closed 2025-09-27 by maxwxyz