FreeCAD/FreeCAD #15992: CAM: MillFace, line pattern, does not respect climb cut mode

sparker0594, 2024-08-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The CAM_MillFace Operation does not respect the Climb Cut Mode in several set-ups:
Using [Dev 38495], Stable 0.21.2 crashed when attempting to replicate.

MillFace, Line Pattern, Climb Cut mode. The tool path shows end-of-line retraction and rapid to the next line, but the next line is the opposite cutting direction resulting in a Conventional Cut Mode.

I would have expected the cutter to rapid retract, then rapid to the start of the next line (initiating the next cut pass from the same side at the previous, offset by the step-over percent)

[image]

My attempt to illustrate what I mean:
[image]

The Current Line toolpath output is similar to the ZigZag pattern, but with retracts between each line:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git)
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.65
  * CurvedShapes 1.0.9
  * Curves 0.6.42
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

CAM

### Anything else?

[Pull Start Adapter Plate.zip](https://github.com/user-attachments/files/16692872/Pull.Start.Adapter.Plate.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-01:
@Russ4262 would you have a moment to confirm ?

Russ4262, 2024-12-07:
Opened test file provided by @sparker0594 and recomputed the MillFace op with no correction to the paths for the 'Line' cut pattern.

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39287 (Git) Conda
Build type: Release
Branch: main
Hash: b5a43eb2208b91feb76b2cc363e3956922c3ad75
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * FC_SU
  * freecad.gears 1.0.0
  * freecad.pathred
  * Z_MacroStartup

3nt3, 2025-04-06:
I just noticed this and it's very annoying because my machine simply doesn't have the rigidity to climb cut in steel. I'll take a look at the code later

bombela, 2025-07-15:
None of the face milling patterns appear to be respecting climb/traditional. They all appear to do a zigag or contour, picking a random cut direction.

closed 2025-11-28 by chennes