FreeCAD/FreeCAD #15994: CAM: MillFace Min Travel selected reverses the Cut Mode with Offset Pattern (Climb to Conventional)

sparker0594, 2024-08-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[Dev 38495]
CAM_FaceMill, when using the Ossfet Pattern, the cut mode will be reversed when the Min Travel option is selected.
Through all of the example GIF's below Climb milling was selected. It can be seen when Min Travel is selected the cutting mode is reversed in the simulation (Climb becomes Conventional).
When the Min travel is selected and Conventional cut mode is selected the cut mode becomes climb in the simulation.

Example, Min Travel is selected, and climb cut mode is selected. The toolpath produced is conventional.

[image]

Example, Min Travel is selected, and climb cut mode is selected.  The toolpath produced is climb.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git)
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.65
  * CurvedShapes 1.0.9
  * Curves 0.6.42
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

CAM

### Anything else?

[Pull Start Adapter Plate.zip](https://github.com/user-attachments/files/16694184/Pull.Start.Adapter.Plate.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

iromero91, 2024-08-22:
I can confirm this, further it seems like it will also make multiple passes alternate between going from outside in to inside out

luzpaz, 2024-08-22:
@iromero91 can you please attach your full [About](https://wiki.freecad.org/About) info that you reproduced this on? TIA

closed 2025-11-28 by chennes