FreeCAD/FreeCAD #15995: Assembly: Cannot constrain to the assembly origin planes/axes

tibfel, 2024-08-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I don't know if this is a bug or a not-yet-developed functionality, but I am unable to constrain anything to the origin planes/axes of the assembly.  In the future, there might be an initiative to add circular/rectangular pattern function into the assembly and I am used to modelling half of the symmetric assembly and then just revolve the other half around the Z axis. 

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git)
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Slovak/Slovakia (sk_SK)
Installed mods: 
  * OpenTheme 2024.8.15
  * Render 2024.8.4
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-21:
@tibfel did you check https://github.com/FreeCAD/FreeCAD/labels/Mod%3A%20Assembly to see if this ticket already exists ?

tibfel, 2024-08-21:
I tried to search open issues with assembly constrain and assembly constraint keywords and did not find anything. Maybe I did a wrong search. Sorry.

luzpaz, 2024-08-21:
CC @PaddleStroke

mno13, 2024-08-21:
similar to:

https://github.com/FreeCAD/FreeCAD/issues/12793

https://github.com/Ondsel-Development/FreeCAD/issues/57

luzpaz, 2024-08-22:
Closing this as a duplicate then ?

PaddleStroke, 2024-08-28:
No it's not the same issue @luzpaz

PaddleStroke, 2024-08-28:
Currently you can't use the assembly own origin object in joints.
It is a feature request. Which is not an easy one. So it won't be for 1.0 I think.

tibfel, 2024-08-28:
The solution could be to place an empty part into the assembly and add constraints to its origin planes. However, I have found out that I can only constrain origin plane of one part to a face of another part. Constraining origin plane of one part to the origin plane of another part is not possible.

Zolko-123, 2024-08-28:
Is it possible to constraint a (link to a) part to a solid in the assembly ? (for example a link to a Body to a Body inside the assembly)

closed 2024-12-13 by chennes